gnuradio/gnuradio #7589: After removing the last-opened flowgraph, a segfault occurs when starting up with `gnuradio-companion`

marcusmueller, 2024-12-26:
Extracted from #6957.

marcusmueller, 2024-12-26:
@lukazlim said:

> 
> The `--gtk` version has no issues, it defaults to a new flowgraph which should be the intended behaviour.
> 
> [image]
> 
> Restoring/moving the flowgraph back to its original directory will stop those errors until the program is closed while on a new unsaved flowgraph.
> 
> Tested on Ubuntu 22.04.4 with Python 3.10.12

haakov, 2024-12-26:
This one needs some investigation. It looks like the segfault is coming from GRC Gtk

marcusmueller, 2024-12-27:
Correct, the Qt version doesn't print the `<<< Welcome to GNU Radio Companion` in the log level setting that @lukazlim seems to use. 

@lukazlim how are chances you would be able to reproduce the issue for us? I'd especially be interested in a *backtrace*, if that would be possible

In short: 

0. install `gdb` if you don't have that, 
1. run `gdb --args "$(head -n1 $(which gnuradio-companion) | sed 's;^[^/]*\(.*\)$;\1;')" "$(which gnuradio-companion)"` <br> (all this does is finding out which python interpreter *exactly* is in use to execute the `gnuradio-companion` python program, then invoking that with the path to the same program), then
2. on that "`(gdb)`" prompt :
    1. enable logging:  `set logging on`
    2. set where to log: `set logging file grc_issue_7589.txt`
    3. run GRC: `run`
    4. wait till segfault
    5. dump all threads' backtraces: `thread apply all backtrace`
    6. quit: `quit`
3. upload the `grc_issue_7589.txt` here