gnuradio/gnuradio #7596: Variable Editor window shows "" for many variable values rather than a calculated value.

marcusmueller, 2024-12-26:
### What happened?

spin-out of #6957

### System Information

a

### GNU Radio Version

3.11-git (main)

### Steps to Reproduce the Problem

GRC-Qt

### Relevant log output

```shell
For example a QT GUI Range block's variable.
```

haakov, 2024-12-28:
Qt GUI Range seems to work, at least for the default case. Are there other examples of values that don't show up properly?

[image]

marcusmueller, 2024-12-30:
The field seems to show literal value, not calculated value. Which seems to be less confusing, and thus the UX-desirable thing. In that, it seems to be reliable, at least the few things I could test. So, closing.

closed 2024-12-30 by marcusmueller