gnuradio/gnuradio #7604: Investigate remote build performance on OSUOSL (webdav/sccache)

marcusmueller, 2024-12-28:
I've set up an experimental host on OSUOSL's openstack. Need to:

- [ ] set up a .gnuradio.org subdomain (@noc0lour probably can tell me how)
- [ ] set up a letsencrypt cert / certbot (or do we have wildcard subdomain certs? need to check)
- [ ] set up [nginx for webdav with user/password auth](https://github.com/ccache/ccache/wiki/HTTP-storage#nginx) 
- [ ] run a GR build with caching on that from home, run a build pulling from that cache from azure cloud

ei8fdb, 2025-01-17:
@marcusmueller you want help with some of this?

marcusmueller, 2025-01-18:
generally, yes, but! I'm currently blocked by some headscratcher of an issue on a feature I just introduced, so it's kind of expected that I can't really do much here, and it's also not very pressing. I could certainly do better documenting my progress here, especially since I'd be bound to forget otherwise!

marcusmueller, 2025-01-18:
Progress:

> set up [nginx for webdav with user/password auth](https://github.com/ccache/ccache/wiki/HTTP-storage#nginx)

I can do that with `ccache` (**not** sccache) as the communicating part. However, `ccache` supports no TLS, and sending plaintext authentication over the internet unencrypted is… let's not even do that, lest we want to become involuntary file hoster for a monetarily motivated, morally uninhibited (surprise) third party.

Also, `ccache` needs that secret to be etiher in `~/.config/ccache/ccache.conf` or in an environment variable; both places that inherently make it easy to exfiltrate secrets, as the user running the CI run needs to be able to read these.

I need to evaluate whether the situation is better with `sccache`, as that might allow us to run the local job handler daemon with knowledge of these auth secrets without giving the user runnning the actual CI access to the secrets.

Barring that, hm, tough! We could actually do something like *not* use basic auth and *not* use TLS, and connect to a TLS-encapsulating and authenticating local proxy *within* the CI (that runs as different user and thus can get the secret). Whether such a thing exists? Don't know.

But that's the thing: auth with secrets in semi-trustworthy environments is  hard. `ccache` itself is not up to the task at all architectually, maybe `sccache` is, but that's bound to be a lot of fun.

marcusmueller, 2025-01-18:
writing this down, possibly a step where we generate a one-time access token and exchange it with the server in the CI "make build" workflow is a solution. (we'd be wary of people modifying the workflow files in PRs, and not approve the PR run if they are new users and do fishy things to the workflow, including uploading secrets to third parties. We then wouldn't have to generally read every line of code submitted for review prior to clicking the "new user. Run CI?" button very closely.)

marcusmueller, 2025-06-19:
with OSUOSL facing financial problems: not sensible at this point