gnuradio/gnuradio #7611: GRC Qt Move actions are duplicated

haakov, 2024-12-29:
### What happened?

When moving a block in GRC Qt, two actions are added to the undo stack. In other words, the user has to hit Ctrl+Z twice to undo the move. It looks like the first action contains the entire move, and the second one is empty. **However**, this only happens if you deselect the block after moving it.

### Steps to reproduce

- Move a block
- Click anywhere in the canvas
- Hit Ctrl+Z once (nothing happens)

### Where to start

Place a breakpoint here:
https://github.com/gnuradio/gnuradio/blob/76900adac6ab09ba3344412589fd590cf02a8ed6/grc/gui_qt/components/undoable_actions.py#L62-L64

Sahil-u07, 2024-12-30:
@haakov  please review the changes. The DCO check has passed and the commit has been amended.

marcusmueller, 2024-12-30:
@Sahil-u07  you don't have to remind maintainers to review stuff – it's not even an hour since you made this pull request!

closed 2025-02-03 by haakov