gnuradio/gnuradio #7612: GRC Qt block library completion needs improvement

haakov, 2024-12-29:
This is a GRC Qt issue, but it's not quite feature parity-related.

When I initially added the QCompleter to the block library, I chose the InlineCompletion mode. The reason for this was that this enabled the user to add an Add Const block like this:

1. Ctrl+F, 
2. write "Add Co"
3. Enter

However, the InlineCompletion mode isn't very pretty. PR https://github.com/gnuradio/gnuradio/pull/7435 replaced the InlineCompletion with PopupCompletion, which looks nicer and cleaner, but now we need to use arrow keys to make use of the completion. Also, it's _sort of_ doing the same thing as the TreeView below. Maybe it would be best if we drop the QCompleter altogether, and rather use the TreeView in the same role? Such that the top result in the TreeView will be selected automatically, and can be added by simply pressing Enter? Also, the up/down arrow keys should act on the TreeView.

@dl1ksv thoughts?

dl1ksv, 2024-12-30:
@haakov I understand your intention and I agree. 
But I would not remove the completer as it does some usefull coding for us. What's missing is the possibility to select and add a block by pressing 'Enter'.
We could use the return_pressed signal of the self._search_bar and connect it to a slot that handles this.
This slot should add the block to the fg, if the selection is unique. If the selection is ambiguous we should take first one in the popup list if there is none highlighted, otherwise take the highlighted.

Two additional remarks
1. At the moment we use the default filtermode Qt::MatchStartsWith, but perhaps the setting Qt::MatchContains fits better ? 
2. I'm missing a button to clean up the the search text field, especially on long text entries. Some times ago I tried to construct such a button, but failed.

dl1ksv, 2024-12-30:
I just saw that there is already a on_return_pressed function. This function should be extended to do the job.

haakov, 2024-12-30:
Good points! I tried this, seems to work alright:

```python
        self.returnPressed.connect(self.on_return_pressed)

    def first_block_in_results(self):
        model = self.parent.completer.completionModel()
        index = model.index(0, 0)
        return model.data(index)

    def on_return_pressed(self):
        label = self.first_block_in_results()
        if label in self.parent._block_tree_flat:
```

1. We're already using MatchContains, are we not?
2. We should definitely have such a button. I originally thought it could be a part of the title bar widget, but maybe its better to have it as a grey "X" at the right end of the QLineEdit?

dl1ksv, 2025-01-01:
You are right, we are already using MatchContains. Sorry , my fault.

I tested your proposed code and found some problems.

I get:  
 File "/usr/local/gnuradio/lib64/python3.12/site-packages/gnuradio/grc/gui_qt/components/block_library.py", line 43, in first_block_in_results
    model = self.parent.completer.completionModel()
            ^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'BlockLibrary' object has no attribute 'completer' 

  This can be easily fixed by replacing this line by 
    model = self.completer().completionModel()

Now the code works, but I run into another problem.
Ctrl + F
write Add 

leads to an popup list that contains

Add Const
Add

and some other entries.
If I now press  enter, the 'Add Const' block is added to the fg, even if I select the 'Add' block by pressing the down arrow key twice.

So I modified the code to :

    `def first_block_in_results(self):
        model = self.completer().completionModel()
        index = model.index(0, 0)
        return model.data(index)

    def on_return_pressed(self):
        block_key = None
        label = self.text()
        if label in self.parent._block_tree_flat:
            block_key = self.parent._block_tree_flat[label].key
        else:
            label = self.first_block_in_results()
            if label in self.parent._block_tree_flat:
                block_key = self.parent._block_tree_flat[label].key
        if block_key:
            self.parent.add_block(block_key)
            self.setText("")
            self.parent.populate_tree(self.parent._block_tree)
        else:
            log.info(f"No block named {label}")
            self.parent.reset()
`

Now, if the entered text is exactly the name of a block, the block is entered on 'Enter', even if the block is not the first in the list.

I'd prefer to change the sort order in the popup, that is 'Add' should be sorted before 'Add Const', but I failed.

I'd prefer a  'clear button'  inside QLineEdit. We could just simply use QLineEdit.setClearButtonEnabled(self,True) in the init of the search bar. But I was not able to catch the 'clear signal' from this button for some further processing. Perhaps catching the editingFinished signal may be helpfull.

PR #7644 (2025-01-21): grc-qt: Some Improvements for the block library completion

## Description

See discussion in #7612

Add a clear button to the searchbar
## Related Issue

Fixes #7612 

## Testing Done

Search for Add or "Add C" , use key up / down keys
## Checklist
closed 2025-01-21 by willcode