gnuradio/gnuradio #7615: Var.-Editor context menu: Be context-sensitive!

marcusmueller, 2024-12-31:
as of 37251f139961c9ff4dfd0039d84593842d7478fd:

[image]

- Show "enable" only when disabled
- Same, but opposite for "disabled"
- Remove "add import" and "add variable", unless right-clicking on the variables or imports category header.
- Properties dialog doesn't open when clicking "Properties…"

Also: 

- clicking on the "+" next to "Variables" and "Imports" is nonfunctional
- clicking on the "-" next to and entry is nonfunctional

Harishankar14, 2024-12-31:
"Hi @marcusmueller, I’d like to take up this issue and improve the context menu behavior and functionality in the Variable Editor. My approach will involve:

Making sure that  the menu options context-sensitive (e.g., showing 'Enable' only for disabled items and vice versa).
Restricting 'Add Import' and 'Add Variable' to category headers only.
Fixing the 'Properties' action to open the dialog correctly.
Addressing the nonfunctional expand/collapse behavior for '+' and '-' buttons.
I'll test all changes thoroughly to ensure everything works as expected. Let me know if there are any additional considerations.

marcusmueller, 2024-12-31:
Awesome, @Harishankar14!

Please go ahead! Looking forward to your pull request (pro trick: you can open a "draft" PR early on with the first modifications you've done, even if things aren't perfect yet).

the-gremlin, 2025-01-04:
Hi @Harishankar14, would you mind if I took on the "+" and "-" buttons issue? I could even help you with reviewing your code or by tackling more of the issue.

Harishankar14, 2025-01-05:
@the-gremlin  go on

ei8fdb, 2025-01-17:
Hi @Harishankar14 is there anything you need/are waiting for with your part (i.e. - the "+" and "-" buttons handled by @the-gremlin ) of this issue?

Harishankar14, 2025-01-19:
@ei8fdb would love to work on this 

- Show "enable" only when disabled

- Same, but opposite for "disabled"

- Remove "add import" and "add variable", unless right-clicking on the variables or imports category header.

- Properties dialog doesn't open when clicking "Properties…"

PR #7709 (2025-02-03): grc-qt: Fix properties dialog in variable editor, alternativ solution
Add the property dialog to the variable_editor contextmenu

Partially fixes #7615

The missing context sensitivity should be handled  in an other pr.

## Description

Add the property dialog to the variable_editor contextmenu
This is an alternative solution for #7690

The missing context sensitivity should be handled  in an other pr.
## Related Issue

Partially fixes #7615
## Testing Done
commit 715da4d1 (2025-02-03): grc-qt: Fix properties dialog in variable editor, alternativ solution

Add the property dialog to the variable_editor contextmenu

Partially fixes #7615

The missing context sensitivity should be handled  in an other pr.
commit dd2b6490 (2025-02-05): grc-qt: Fix properties dialog in variable editor, alternativ solution

Add the property dialog to the variable_editor contextmenu

Partially fixes #7615

The missing context sensitivity should be handled  in an other pr.
closed 2025-02-03 by willcode