gnuradio/gnuradio #7617: make grc buildable without gtk deps

Hoernchen, 2025-01-03:
### Feature Description

pure 3.10 qt builds on windows are fairly easy, the only problem is the GI deps for gtk. Just removing the gtk check from the companion and the 4 gi deps from the cmake file is sufficient to build and use the qt version, so those deps should probably be optional.

### Feature Urgency

medium (would be nice to have in the near future)

marcusmueller, 2025-01-03:
Considering GRC still starts in GTK mode by default: For Qt||Gtk to be essentially an "OR" dependency for GRC, we need GRC to use Qt (without `--qt` being specified at the command line or it being configured to) when Gtk isn't available. That sounds doable.

haakov, 2025-01-06:
It sounds doable, we _don't_ already do it, and I agree that we should do it 😄

marcusmueller, 2025-01-07:
The python part is easy: simply, when there's no explicit setting of a flavor, try `run_gtk`, and if that fails with an import error, `run_qt`. 

The annoying part as usual is CMake. How about this strategy for grc/CMakeLists.txt:

We introduce two new components, 

- `GR_REGISTER_COMPONENT("grc-qt" ENABLE_GRC_QT ${grc_qt_deps})`  and 
- `GR_REGISTER_COMPONENT("grc-gtk" ENABLE_GRC_GTK ${grc_gtk_deps})`,

add a new variable `ANY_GRC_ENABLE` that's true if either of `ENABLE_GRC_{QT,GTK}` is true,
and modify the `"gnuradio-companion"` component so to depend on that, and the common python dependencies?

haakov, 2025-01-08:
Sounds like a reasonable strategy. What if neither Qt nor Gtk is found, can we still run `grcc`?

dl1ksv, 2025-01-13:
This does not solve the build problem , but if you start once
grc --qt 
and select in the preference option GRC QT as 'Default  Grc Version' , ( don't forget to save ) then the qt version of grc is started next time if you run grc