gnuradio/gnuradio #7620: block::declare_sample_delay: public, but not threadsafe (nor documented to not be)

marcusmueller, 2025-01-04:
### What happened?

Investigating some modem misbehaviour, it rose to my attention that we're not taking care that public setters don't interfere with things like tag pruning.

### System Information

n/a

### GNU Radio Version

3.11-git (main)

### Specific Version

v3.11.0.0git-862-ge95299ec5

### Steps to Reproduce the Problem

n/a

Harishankar14, 2025-01-05:
How does this implementation impact tag pruning functionality? Specifically, does it lead to inaccurate tag propagation or the loss of tags?

marcusmueller, 2025-01-05:
@Harishankar14 no, it currently doesn't affect tag propagation negatively, because we take care to only call the function in a context where the modification mutex is already held.

However, this function can be called, as it is public, without holding the mutex, and thus, it's an easy way to cause very hard to understand bugs.

Harishankar14, 2025-01-06:
Would it make sense to add a check inside the function to ensure the mutex is held before making modifications? That way, we can catch potential misuse early, even if it adds some development-time overhead.

marcusmueller, 2025-01-06:
"someone is holding the mutex protecting these data structures that I'm about to modify" is sadly not helpful in figuring out whether the usage is legitimate