gnuradio/gnuradio #7624: annotator_raw: d_queued_tags is std::vector, gets sorted after every insertion, undergoes many element removals: wrong container

marcusmueller, 2025-01-05:
### What happened?

<meta>This is a **good first issue** for people who already know C++ and can figure out how to build GNU Radio on their machine.</meta>

checking the abuse of `std::sort` on tag vectors, I came across this. Sorting a vector every time you insert something screams "use a sorted container, not a `std::vector`".

Also, going forwards (n=0,…,N-1) through a N long vector and erasing the n. element causes N-1 copies of length N-1-n to happen, which happens to be in the order of N² copied elements… uhhhh…

In this case, a `std::set d_queued_tags` would do. 

Also, replace the `while(i!= d_queued_tags.end())` loop with a C++11 `for(const auto& tag: d_queued_tags)` loop.

The block has no unit test. This is also not good.

### System Information

n/a

### GNU Radio Version

3.11-git (main)

### Specific Version

f3127fce18f2363c978af63aec686a72da4cb042

### Steps to Reproduce the Problem

n/a

Mindslayer001, 2025-01-09:
Hi @marcusmueller, I’d like to work on this issue and would be thrilled to do so under your guidance again.

Mindslayer001, 2025-01-09:
this my changes according to the pr.

`gr-blocks/lib/annotator_raw_impl.cc`
before:
```c++
// locate queued tags that fall in this range and insert them when appropriate
    std::vector<tag_t>::iterator i = d_queued_tags.begin();
    while (i != d_queued_tags.end()) {
        if ((*i).offset >= start_N && (*i).offset < end_N) {
            add_item_tag(0, (*i).offset, (*i).key, (*i).value, (*i).srcid);
            i = d_queued_tags.erase(i);
        } else {
            break;
        }
    }
```
after:
``` c++
 for (auto it = d_queued_tags.begin(); it != d_queued_tags.end(); ) { //changed accord to the pr
        const auto& tag = *it;
        
        // Check if the tag falls within the desired range
        if (tag.offset >= start_N && tag.offset < end_N) {
            add_item_tag(0, tag.offset, tag.key, tag.value, tag.srcid);
            it = d_queued_tags.erase(it);
        } else {
            // Break out of the loop if the tag is out of the range
            break;
        }
    }
```

`gr-blocks/lib/annotator_raw_impl.h`
before: 
```c++
class annotator_raw_impl : public annotator_raw
{
private:
    const size_t d_itemsize;
    std::vector<tag_t> d_queued_tags;
    gr::thread::mutex d_mutex;

public:
    annotator_raw_impl(size_t sizeof_stream_item);
    ~annotator_raw_impl() override;

    // insert a tag to be added
    void add_tag(uint64_t offset, pmt::pmt_t key, pmt::pmt_t val) override;

    int work(int noutput_items,
             gr_vector_const_void_star& input_items,
             gr_vector_void_star& output_items) override;
};
```

after: 
```c++
class annotator_raw_impl : public annotator_raw
{
private:
    const size_t d_itemsize;
    std::set<tag_t> d_queued_tags;
    gr::thread::mutex d_mutex;

public:
    annotator_raw_impl(size_t sizeof_stream_item);
    ~annotator_raw_impl() override;

    // insert a tag to be added
    void add_tag(uint64_t offset, pmt::pmt_t key, pmt::pmt_t val) override;

    int work(int noutput_items,
             gr_vector_const_void_star& input_items,
             gr_vector_void_star& output_items) override;
};
```
`gnuradio-runtime/include/gnuradio/tags.h`
before:
```c++
#include <gnuradio/api.h>
#include <pmt/pmt.h>
```
after:
```c++
#include <gnuradio/api.h>
#include <pmt/pmt.h>
#include <set>
```

marcusmueller, 2025-01-09:
hi @Mindslayer001  can you move this to a PR? (you said "my changes according to the PR, but I don't see the PR :) ). Also, first review while skimming this: did you take care of 

> Also, replace the while(i!= d_queued_tags.end()) loop with a C++11 for(const auto& tag: d_queued_tags) loop.

?

PR #7749 (2025-06-07): blocks/annotator_raw: use set instead of vector to avoid sorting
## Description

`annotator_raw` used a vector as queue for randomly inserted tags (i.e., things were re-sorted all the time). Changed that.

Worse, honestly, is that it had no test coverage. Changed that, too.

## Related Issue

Closes #7624

I thought #7624 was a good beginner's issue to fix, especially with how much work I put into describing the necessary changes. It seems it wasn't, really. Sigh.

## Which blocks/areas does this affect?

annotator_raw

## Testing Done

added new unit test that also covers the temporal aspects [... cut]
closed 2025-06-07 by marcusmueller