gnuradio/gnuradio #7665: USRP source and sink block doesnot honur the cmd.start/stop command

sandeep-bastola, 2025-01-16:
### What happened?

I have created a custom gate to command module where I input gate signal and based on 1 or 0 in the signal, the command to start or stop the USRP source or sink is sent. I can see from logs that the commands cmd.start/stop are created and send to the command port of USRP source. But it doesnot have any effect on the output of the USRP source block. Ideally when I stop the USRP source from cmd.stop command, it should not receive any signal and show no signal on QT time sink. But it is always showing some received signal. Anyone has faced similar situation and or fixed it ? Any help will be highly appreciated

### System Information

OS: Mac
GR Installation Method: conda
GNU Radio version: 3.10.11.0
UHD version: 4.7.0.0

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.11.0

### Steps to Reproduce the Problem

create a gate to command custom block to send cmd.start signal while gate signal is 1 and cmd.stop signal when gate signal is 0. 
Feed the output of the custom block to the command port of USRP

### Relevant log output

```shell

```

ei8fdb, 2025-01-17:
Hi @sandeep-bastola I'm helping GR withGH issue responses.

Can you tell me if this open PR relates to your issue: https://github.com/gnuradio/gnuradio/pull/7668

Thanks/bernard

willcode, 2025-01-17:
I think the feature request is for a separate block that acts as a translator between the value of a stream and UHD commands, sort of like a squelch. The PR is a bug fix.

sandeep-bastola, 2025-01-17:
I don't think this PR is related to my issue. This PR mentioned about replay block, but the issue I have is with USRP sink / source block. 

> Hi [@sandeep-bastola](https://github.com/sandeep-bastola) I'm helping GR withGH issue responses.
> 
> Can you tell me if this open PR relates to your issue: [#7668](https://github.com/gnuradio/gnuradio/pull/7668)
> 
> Thanks/bernard

willcode, 2025-01-18:
I read this wrong the first time. The two may be related. The PR is merged `main`, so if able, see if that helps.

mbr0wn, 2025-01-30:
- The USRP source block does not have a 'start' and 'stop' command (unlike the replay block).
- Note that the RX streamer block has a stream_cmd command. We could consider adding that to the USRP source.

Overall, this is a feature request, but not a bug report.