gnuradio/gnuradio #7669: QT GUI Dial -- quantized increments

roberthadow, 2025-01-16:
### Feature Description

Dial is a beautiful user interface.  It would have increased utility were it to have the option of quantized output.  It would be nice to be able to select a specific increment.  One is great, but I'd like to see 10, 100, and 200,000. 

Use case: the dial is used to select an FM station in the USA. As long as the minimum and maximum are selected properly, an increment of 200k would land the user exactly on the FM frequencies she or he desires. Other bands require other increments.

A nice-to-have would be the accomodation of scientific notation throughout the feature, as long as the chosen number is an integer.

### Feature Urgency

low (just an idea)

ei8fdb, 2025-01-17:
Hi @roberthadow thanks for opening this issue. I'm helping GR with responses to GH issues. To make sure I understand you right:

You're describing the ability to define an increment step for the dial to use when the flow graph runs. In your broadcast FM example:

with a step of 200KHz tuning between e.g. min 88.0MHz and max 90.0MHz, the dial would tune from 88.0 to 88.2, 88.4........to 90.0MHz. Correct?

You'd also like to use scientific notation to define the increment. Right?

Where would you expect to define that increment? In the dial block itself, or somewhere else? Would you expect to use that increment in GR in the commandline also? Or just in GRC?

Thanks / bernard

roberthadow, 2025-01-19:
Hello Bernard --

I use the GRC almost exclusively.  Whether this idea fits on the command line, I don't know.

QT GUI Range has three means for the user to interact.  Using a slider on a range of millions of possibilities is quite useless and not visually appealing. QT GUI Range has a step attribute.  For consistency, in QT GUI Dial the increment should also be called "step."  When one clicks the up or down arrows, the value increments or decrements by the step value. That's perfect.  Or the user can type a specific value. I suggest QT GUI Dial block work the same way.

To be consistent with all other features, yes, one should be able to use scientific notation for all attributes.

Your understanding of everything else is spot on.

marcusmueller, 2025-01-19:
Hi! Chiming in from the technical side: You're right, GUI elements by themselves don't have a command line, but you could build such a thing using a "Parameter" block in GNU Radio, and using the ID of that block as variable name that you could put into the "step" field. Then, you could run your generated flow graphs with something like `./roberts_flowgraph.py --option-name-i-chose 42` to set the step size via command line :)

marcusmueller, 2025-01-19:
Regarding scientific notation: all option fields in GRC right now support "normal" Python. So, as it is right now, we can support entering `200000`, or `2*10**5` or `200e3`, but also things like `value_from_parameter * 1000` or `math.sin(2*math.pi*value_from_parameter)`.

Sadly, this flexibility means that we can't directly support `200k` or `200kHz`; that would require both relatively involved changes to the parsing logic in GRC, as well as a compatibility-breaking change of the GRC file formats, to store the information whether this should accept (and show and mean) scientific notation or Python.

I'd hence recommend splitting this into the "very achievable" part of "add a step option to the dial" and the far harder "make GRC fields able to accept *either* Python *or* scientific notation" (sadly, the two are mutually incompatible), and how to present that to the user, and how to make the way the input field currently works visible to them (and whether to do it at all; I must admit having a field that can either accept `200k` or `1e3*value_from_parameter`, but never `1k*value_from_parameter`, that's going to be very confusing to me at least. Problem really is that scientific notation clashes with how Python is defining where the boundaries between number literals and names of variables, operators and functions are).

marcusmueller, 2025-01-19:
😄  ! 

by the way, Python accepts all `200e3`, `200E3`, `200e+3`, `200E+3`, so you really can keep R compatibility in your notation if you will; https://docs.python.org/3/reference/lexical_analysis.html#floating-point-literals documents the valid ways you can enter floating point numbers in Python.

(Context for future software-design focused readers: things get slightly complicated when you try to support all the ways C++ can read floating point literals; `0xDEAD.CAFEp00` is a valid C++ floating point literal, equal to 57005.79296875 in case you wonder. And there's the thing where things get a little strange, because you can generate C++ or Python flowgraph programs out of your GRC flowgraph. We simply take what you entered in a field literally and put it in the generated source code; however, GRC is a python program, so what you enter in the fields gets interpreted as python for design purposes, but gets interpreted as C++ for compilation purposes when you select to build a C++ flow graph. Talk about confusing users… we'd all love to fix that, but it's a larger development effort that would be necessary to a) be able to use a C++ compiler to parse these expressions and b) to give that C++ parser a view of all the names that are defined through the blocks you use in your flow graph, so it could do its job on `std::sin(2*std::numbers::pi*value_from_parameter)`)