gnuradio/gnuradio #7672: PMT de-serialization from a file not working correctly

ptrkrysik, 2025-01-21:
### What happened?

PMT's content obtained after de-serialization from a file is not the same as before serialization of a PMT to a file.

### System Information

OS: Ubuntu 24.04
GR Installation Method: Source

### GNU Radio Version

3.11-git (main)

### Specific Version

v3.11.0.0git-848-g6fd69852

### Steps to Reproduce the Problem

Here is simple test code (compile it with 'g++ test.cpp -lgnuradio-pmt -o test'):
```
#include <iostream>
#include <sstream>
#include <pmt/pmt.h>
#include <fstream>

void test_pmt_serialization() {

    pmt::pmt_t pdu_header = pmt::make_dict();

    uint8_t buf[10];
    for (int i = 0; i < 10; i++) {
        buf[i] = i + 1;
    }

    pmt::pmt_t blob = pmt::make_blob(buf, 10);
    pmt::pmt_t msg = pmt::cons(pdu_header, blob);

    std::ofstream output_file("out", std::ofstream::binary);

    pmt::serialize(msg, *output_file.rdbuf());
    output_file.close();

    std::ifstream input_file("out", std::ifstream::binary);

    pmt::pmt_t msg_deserialized = pmt::deserialize(*input_file.rdbuf());

    std::cout << "Before serialization:   " << msg << std::endl;
    std::cout << "After de-serialization: " << msg_deserialized << std::endl;

    input_file.close();
}

int main() {
    test_pmt_serialization();
    return 0;
}
```

After running the result printed on the screen is following:
```
Before serialization:   (() . #[1 2 3 4 5 6 7 8 9 10])
After de-serialization: (() . #[7 8 9 10 0 0 0 0 0 0])
```

Expected is to get the PMTs with the same content (before serialization and after de-serialization) but they are different.

The issue causing bug in GNU Radio's OOT described in issue #6587.

### Relevant log output

```shell

```

ptrkrysik, 2025-01-21:
The issue was introduced by change to the 'pmt_serialize.cc' file by this commit c1c56d2d149fa1663e0d41b7f50ad60f67258ce9.

willcode, 2025-01-21:
File contents out output file are, so the full vector is there:
```
0000000  07  06  0a  00  00  00  00  0a  01  00  01  02  03  04  05  06
0000010  07  08  09  0a                                                
```

willcode, 2025-01-21:
@jacobagilbert could you take a look at this to see if it's something simple like a type conflict?

ptrkrysik, 2025-01-22:
In case of the above test code the issue is caused by this single change from:
```C++
static bool deserialize_untagged_u32(uint32_t* ip, std::streambuf& sb)
{
    std::streambuf::traits_type::int_type t;
    int i;

    t = sb.sbumpc();
    i = t & 0xff;

    t = sb.sbumpc();
    i = (i << 8) | (t & 0xff);
    t = sb.sbumpc();
    i = (i << 8) | (t & 0xff);
    t = sb.sbumpc();
    i = (i << 8) | (t & 0xff);

    *ip = i;
    return t != std::streambuf::traits_type::eof();
}
```
to
```C++
static bool deserialize_untagged_u32(uint32_t* ip, std::streambuf& sb)
{
    std::streambuf::traits_type::int_type t;
    t = sb.sgetn((char*)ip, sizeof(uint32_t));
    sb.pubseekoff(sizeof(uint32_t), std::ios_base::cur);
    *ip = be32toh(*ip);

    return t != std::streambuf::traits_type::eof();
}
```

It is probable that the same problem is also with other data types supported by different 'bool deserialize_untagged_*' functions, as the changes applied to them seem very similar.

willcode, 2025-01-22:
Thanks for tracking that down!

ptrkrysik, 2025-01-22:
I suppose the difference is in what happens to 'sb' parameter in the function.

ptrkrysik, 2025-01-22:
This already shifts the streambuf internal position:
    t = sb.sgetn((char*)ip, sizeof(uint32_t));
so additional shift shouldn't be performed and this should't be done:
    sb.pubseekoff(sizeof(uint32_t), std::ios_base::cur);

That behaviour of 'sgetn' function is not very well documented. There's no mention of it on cplusplus.com or en.cppreference.com, but I've found one here:
https://www.tutorialspoint.com/cpp_standard_library/cpp_stringbuf_sgetn.htm

PR #7682 (2025-01-29): PMT: Remove detrimental calls to 'pubseekoff' from de-serialization

## Description
'pubseekoff' method was used probably with intention to move the std::streambuff's internal position into the right place. However this was already done by 'sgetn' method. With both calls: to 'sgetn' and to 'pubseekoff', the internal position of the 'sb' streambuff ended up being wrong.

## Related Issue
Fixes #7672

## Which blocks/areas does this affect?
Affects all blocks that use pmt::deserialize(...) function. I.e. these blocks from gr-zeromq:
- pull_msg_source,
- req_msg_source,
- sub_msg_source.

## Test [... cut]
closed 2025-01-29 by willcode