gnuradio/gnuradio #7695: tcp_source (from the GRC blks2 time) still bad, still not replaced

marcusmueller, 2025-01-28:
### What happened?

A problem of a user yesterday reminded me of our tcp source still being bad (Details: https://github.com/gnuradio/gnuradio/issues/1540). When we decommissioned the blks2 hierarchy (which where Python-constructed blocks that shipped with GRC instead of any of the gr-* modules), we just moved the `tcp_sink` to gr-network. (Although we knew it was bad.)

We need to replace it with one that

- doesn't wait for a connection in the constructor, meaning that you can't even *construct*, leave alone *start* a flow graph without a connection being made (and that can, and does, lead to deadlocks)
- has unit tests for all connection directions and starting orders
- is C++
  - has GRC C++ generation
- correctly handles connection loss

desirable:

- `recv`es directly into output buffer
- allows for re-connection (if requested at constructor time)
- allows for clamping the number of bytes to be received in one call
- avoids dependendency on Boost, since this should all be synchronous anyway (sources blocking in work)  and afaict socket.h works on windows just as well (needs verification)

### System Information

n/a

### GNU Radio Version

3.11-git (main)

### Specific Version

v3.11.0.0git-892-g6406da8f68

### Steps to Reproduce the Problem

n/a

### Relevant log output

```shell

```

marcusmueller, 2025-01-28:
Tagging this bug since complete lack of QA for a block is one in itself; but also, the "needs to have a connection constructor time" is counter-intuitive and a source of application deadlock.

the-gremlin, 2025-06-27:
I will attempt to _very slowly_ work on this issue. Probably will start with a non-templated version first and then we'll see how it goes :)