gnuradio/gnuradio #7696: gr-digital CRC Check default for 'Discard CRC' is 'False'

duggabe, 2025-01-28:
### What happened?

The default for 'Discard CRC' is 'False', which leaves the CRC in the otherwise valid output data. This is not the case for the stream version 'Stream CRC32' in the 'check CRC' mode which works as expected.

### System Information

OS: Ubuntu 24.04.1
GR Installation Method: distro package - apt

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.9.2

### Steps to Reproduce the Problem

pkt_53.grc in https://pastebin.com/0PYkkVUS was created to test this problem. The log output shown below contains the correct payload data "GNU Radio", but also contains the valid CRC. This is neither expected nor desired.

Note that this is not the case for the stream version 'Stream CRC32' in the 'check CRC' mode which works as expected.

### Relevant log output

```shell
***** VERBOSE PDU DEBUG PRINT ******
()
pdu length =         13 bytes
pdu vector contents = 
0000: 47 4e 55 20 52 61 64 69 6f f1 09 aa 8f 
************************************
```

marcusmueller, 2025-01-28:
Reproduction gist:
https://gist.github.com/marcusmueller/28bfc2ded06d85016b46f18b19df41ce

[image]

marcusmueller, 2025-01-28:
Is this a change from previous behaviour of the same block? I ask because this is an explicit setting:

[image]

duggabe, 2025-01-28:
From https://github.com/gnuradio/gnuradio/blob/main/gr-digital/grc/digital_crc_check.block.yml
-   id: discard_crc
    label: Discard CRC
    dtype: bool
    default: False

daniestevez, 2025-01-30:
As far as I know (and git blame agrees), the default has always been false, which also matches the [C++ constructor default value](https://github.com/gnuradio/gnuradio/blob/main/gr-digital/include/gnuradio/digital/crc_check.h#L61).

The default values of CRC Check and CRC Append are intended to match, so that they can be combined in simple examples (such as the one Marcus showed in https://github.com/gnuradio/gnuradio/issues/7696#issuecomment-2619229390) without even requiring the user to understand all these parameters.  For this, I chose the most common CRC-32 as a default, but not for any concrete reason. I could have equally chosen a CRC-16 instead, or CRC-32C. The fact that this CRC-32 matches the one used in the Stream CRC32 block is more of a lucky coincidence than a deliberate effort to make the blocks similar or compatible.

Removing or not removing the CRC after a successful check is a matter of convenience and is related to what kind of processing (if any) comes after the CRC Check. I would say there is no "correct" or "expected" behaviour here. The user needs to choose what they want. I would agree that the documentation could be "buggy" if it fails to point this out, but to me the fact that there is a field called "Discard CRC" plus the extra information [in the wiki](https://wiki.gnuradio.org/index.php?title=CRC_Check) "Discard CRC: A boolean that selects whether the CRC should be removed from the output PDU." seems sufficient.

duggabe, 2025-01-30:
I can fix this in the CRC Check wiki page, but as you can see from my test `pkt_53`, a user might assume (as I did) that the defaults would work without change, i.e. message in => message out (not message out plus CRC).

marcusmueller, 2025-01-30:
I'm confused here a bit, why would a user assume that?

duggabe, 2025-01-30:
> I'm confused here a bit, why would a user assume that?

Why would they not? If I'm sending a message, I expect the ultimate received result to be that message without any left-overs from the transport internals.

marcusmueller, 2025-01-30:
Hm, I'd agree if this block was called a "deframer" or "CRC check & discard", but it's just called "check", but I do see your point, Barry. To me it's just that the message including the CRC is just ever so slightly more likely to be what I want at the output of this, but I'd honestly check each time whether the block I use is configured appropriately.

I don't have a strong opinion on this. Since changing the default in the GRC block definition doesn't break any existing flow graph, we can change that default if anyone has a strong opinion on this, but so far I'd say we have an equilibrium of people thinking each operation is a sensible default. And as such, I'd tend to "do not change behaviour to surprise people already familiar with behaviour unless really necessary".

But I do have a compromise to propose: Barry is right that this is a "fundamental" switch in behaviour, which can be very frustrating if missed, and Dani is right that one would always want to manually configure it to do the right thing anyways.

So, let's make it generally visible in the block, just like the polynomial. That surprises nobody, and makes the step of "oh I should set this thing to behave like I want it to" very clear.

daniestevez, 2025-01-30:
> So, let's make it generally visible in the block, just like the polynomial. That surprises nobody, and makes the step of "oh I should set this thing to behave like I want it to" very clear.

That seems a great idea.

duggabe, 2025-01-31:
I added a note to the 'Discard CRC' parameter and revised the Example Flowgraph to put a note under the CRC Check block.

I am satisfied with this solution.