gnuradio/gnuradio #7697: tcp_source has no QA

marcusmueller, 2025-01-28:
### What happened?

The GNU Radio network.tcp_sink has no QA. 

- There's a comparable qa_tcp_source.py in gnuradio/gr-network/python/network . Copy and modify to check `tcp_sink` works in all of the following scenarios
  - Works in `server=False` mode connecting to a socket, opened at a port >> 1024 by the test case, receives the exact bytes sent through that socket, into a `vector_sink_b`.
  - Works in `server=True` mode, by starting a separate thread in the QA that sleeps 0.1s, then tries to cnnect to a port, which the the `tcp_sink` listens on. Then, verify data as above.
  - End to end `tcp_sink` -> `tcp_source` in two separate flow graphs, checking data, with both
    - sink being the server,
    - source being the server

### System Information

OS: (Linux Distro, Mac, Windows)
GR Installation Method: (native distro packages - apt, rpm, etc., Conda, Source

### GNU Radio Version

3.11-git (main)

### Steps to Reproduce the Problem

n/a

### Relevant log output

```shell

```

Harishankar14, 2025-01-29:
I'd like to work on adding QA tests for tcp_sink. I'll create test cases covering:

Client mode (server=False) with explicit socket connection verification

Server mode (server=True) with delayed client connection

End-to-end validation with tcp_sink → tcp_source in both server/client configurations

I plan to submit a PR within the next 7 days. Let me know if there are additional scenarios to consider

Harishankar14, 2025-02-02:
@marcusmueller.
1. Server as tcp_sink and Client as tcp_source: Where tcp_sink acts as the server, and tcp_source (in client mode) connects to it and receives the data.
2. Server as tcp_source and Client as tcp_sink: Where tcp_source acts as the server, and tcp_sink (in client mode) connects to it and receives the data.

The task mentions that we need to check both scenarios, but I wanted to clarify whether these should be tested separately or if both can be run together in a single flow graph.

Could you please confirm the intended approach? Should we execute the tests in isolation or combine both flow graphs (with tcp_sink and tcp_source in different roles) in a single testing setup?

marcusmueller, 2025-02-06:
Hey @Harishankar14, any update?

Harishankar14, 2025-02-06:
@marcusmueller first two test case are done, having some issue, in the third one, will fix it !

marcusmueller, 2025-02-08:
@Harishankar14 great! as said, draft PRs are more than welcome.

Harishankar14, 2025-02-09:
@marcusmueller yep,created a DRAFT PR!

AB-at-UCR, 2025-03-28:
Hey @marcusmueller, is this issue still open? If so, I would like to work on it.

Gowri213, 2025-12-24:
@marcusmueller Understood, Thank you for your time and clarification.