gnuradio/gnuradio #7698: Incorrect data on X axis in Vector Sink if changed during runtime

mickeyreg, 2025-01-29:
### What happened?

Hi,

A bug happened!

The sample data below is not very useful, but it allowed me to understand where the problem is. Unfortunately I can't fix it myself :(

I wanted to make a Vector Sink whose content changes when the vector length changes. It was an FFT result of varying size, but below is an example that is just to illustrate the problem.

Vector Sink block settings:

[image]

After starting, everything is fine (4096 points from 0 to 4095):

[image]

But when I change the size of the vector:

[image]

The "x-Axis" should be now from 0 to 1024.

From what I've observed: The "X-Axis Step Value" is changed correctly during runtime, but the drawn axis always has the same number of points as it initially had (4096 in this example). The data inside the block changes as intended.

Regards,
Mickey

### System Information

OS: Windows
GR Installation Method: radioconda package

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.10.0

### Steps to Reproduce the Problem

I think it can be recreated if everything is done as described above.

### Relevant log output

```shell

```

willcode, 2025-01-29:
Parameters that are underlined will change at runtime (Vector Size is not).

The request is to make Vector Size and updatable parameter.

mickeyreg, 2025-01-29:
Ok, I understood. The problem is not in the **QT GUI Vector Sink** block, but in the whole diagram. The _Vector Length_ value is not runtime adjustable on other blocks too: **Stream to Vector**, **FFT** (_FFT Size_ value) and others. The problem was visible in the Vector Sink block, but all results were generated for the initial vector length despite modifying the variable.

Thanks for the clarification. I hope that someday this value will be runtime adjustable.

marcusmueller, 2025-01-30:
> Thanks for the clarification. I hope that someday this value will be runtime adjustable.

This is sadly contrary to GNU Radio 3 architecture, where the vector size is constituent to the buffer used between blocks and buffers cannot be reallocated on the fly. Doing that on the fly would also require some mechanism for downstream blocks to know that they need to deal with data differently. For example, the 15. item in the output of a 16-FFT means something different than the 15. item in the output of a 1024-FFT.

Since changing this would require a rather complete redesign of GNU Radio, I'm afraid this won't come to pass.

That doesn't mean there aren't ways to write applications that use "packets" of samples of different lengths, but that usually boils down to *not* going for vectors (but things like streams of individual samples with tags marking the length of each block, or to switch to PDUs exchanged via message passing, not via streams).

closed 2025-01-30 by marcusmueller