gnuradio/gnuradio #7708: Qt6 Port Required

Kangie, 2025-01-30:
### Feature Description

GNU Radio (specifically gr-qtgui) is not compatible with Qt6, and downstreams are beginning to drop
support for Qt5.

https://github.com/gnuradio/gnuradio/blob/914ca6c26d1f42ad170f3b7d48b2a5fa50368b42/gr-qtgui/CMakeLists.txt#L13-L18

The closest thing that I've been able to find to a ticket for Qt6 support in
this repo is #4578 which was resolved by removing Qt6 from the system.
This is not something that can sustainably be maintained going forward.

Downstream Bug: https://bugs.gentoo.org/948049

### Background

https://www.qt.io/blog/qt-offering-changes-2020

Qt5 upstream (Qt Company) OSS support ended on 2020-12-08. Since then, bugs 
are only fixed if reproduced in Qt6 first, then backported. Since then, public 
availability of commercial 5.15 LTS releases is delayed by 1 year, including 
repository access for cherry-picking - this is important, see below, and 
affects Gentoo first and foremost.

The last official LTS release will be this April/May:
https://www.qt.io/blog/qt-5.15-extended-support-for-subscription-license-holders
https://wiki.qt.io/Qt_5.15_Release#Release_Plan

Public availability of this release will be *1 year later*. Some may think 
now, fine, updates (3 in total) until 2026! However, if a dependency/toolchain 
upgrade down the stack breaks Qt5 between now and March/April, to be then 
fixed in their final release, you will be waiting until April/May 2026 for 
that to be available. If it breaks later than April 2025, you will be waiting 
forever. In either case, you rely on downstream volunteer patchwork.

KDE Qt5PatchCollection ultimately relies on Qt company's upstream commits as 
well, so don't expect any original work from there to get things fixed. In any 
case, there is no clear EOL date for these patches, but they *will* be drying 
up as there is no main KDE consumer left.
https://community.kde.org/
Qt5PatchCollection#For_how_long_will_this_be_maintained?

Gentoo Qt Policy page:
https://wiki.gentoo.org/index.php?title=Project:Qt/
Policies#Handling_different_versions_of_Qt

Gentoo Qt migration notes:
https://wiki.gentoo.org/wiki/Project:Qt/Qt6_migration_notes

### Feature Urgency

high (impacts broad portion of ecosystem)

marcusmueller, 2025-01-30:
>  In either case, you rely on downstream volunteer patchwork.

just to be clear here, we're volunteers doing volunteer patchwork :)

marcusmueller, 2025-01-30:
> The closest thing that I've been able to find to a ticket for Qt6 support in
this repo is https://github.com/gnuradio/gnuradio/issues/4578 which was resolved by removing Qt6 from the system.
This is not something that can sustainably be maintained going forward.

to mitigate any panic there: this totally works. People are running GRC-Qt with Qt6  while using Qt5 gr-qtgui widgets on the same machine.

Kangie, 2025-01-31:
> to mitigate any panic there: this totally works. People are running GRC-Qt with Qt6 while using Qt5 gr-qtgui widgets on the same machine. (CC'ing our GRC-Qt lead [@haakov](https://github.com/haakov) on this, in case he wants to add any caveats)

Just noting that I did search for Qt6 before logging this ticket :)

Gentoo (among other distros) are actively working towards removing Qt5, this ticket is a result of that - at some point even this will no longer be suitable (though we're not going to drop Qt5 next week or anything shocking like that); it's more of an FYI for upstreams that are on Qt5 at the moment so that we can get software ported across.

> Do you happen to have any experience doing things like just regex-replacing `qt5` with `qt6` and seeing how much of gr-qtgui breaks?

I'll add it to my TODO; no promises when I get around to checking it though - few other simultaneous big projects to finish off.

maitbot, 2025-02-01:
More things to be aware of...

As Python versions bump up in Debian, troubles in pyside6 crop up
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1092529

I'm installing python3-qtpy-pyqt6 and seeing if the QtPy approach might be best
for gnuradio going forward. Help welcome.

Also, help needed fixing failing tests in pygccxml 2.6.1
(See https://buildd.debian.org/status/fetch.php?pkg=pygccxml&arch=all&ver=2.6.1-1&stamp=1737431080&raw=0 )
I see some new commits since them - I'll update soon and see what gets better.

Alas it's been a long time since qwt got updated in Debian, and an update of qwt is necessary to get qwt working with Qt6.
(gnuradio filter design does some plotting with libqwt.)

Lots to do and a Debian trixie release freeze is coming soon.
-Mait

marcusmueller, 2025-02-01:
Thank you both!

haakov, 2025-02-03:
We've used QtPy for GRC-Qt, so the "Qt6" string probably doesn't show up much in our code. I'd be happy to work on this, it shouldn't take too long to fix. (not sure about qwt though, but I'm sure we'll figure it out)

@maitbot I'll send you a dm

haakov, 2025-02-18:
We've made some progress:
https://github.com/haakov/gnuradio/tree/qt5to6

Shouldn't be too much remaining

argilo, 2026-08-16:
Just an FYI, GNU Radio is deprecated in Homebrew due to the upcoming qt@5 removal scheduled for 2027-05-19.

smunaut, 2026-08-19:
Any news on this ? I see the PR but no activity on it for some time now.

mikemorris, 2026-11-16:
It looks like this was (mostly? fully?) completed and merged in https://github.com/gnuradio/gnuradio/pull/7757 - is a v3.11 release planned anytime soon including this update @marcusmueller? It looks like Homebrew is [planning to drop the gr-qtgui module from gnuradio](https://github.com/Homebrew/homebrew-core/pull/283837) from their build in a few days...

/cc @cho-m @haakov