gnuradio/gnuradio #7712: unable to convert a QVariant back to a Python object

bczhc, 2025-02-04:
### What happened?

Just tried the new GRC Qt GUI, and when I deleted a block, GRC crashed.

### System Information

OS: Arch Linux rolling; kernel 6.12.9-bczhc-dirty
GR Installation Method: From Arch repo/via `sudo ninja install` (bug encounters with both)
python3 GRC uses: 3.13.1
qtpy: 2.4.2

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.11.0

### Steps to Reproduce the Problem

1. Start GRC with `--qt`
2. Drag a random widget
3. Press Delete or Ctrl-X
4. It crashes.

### Relevant log output

```shell
~ ❯ gnuradio-companion --qt                                                                                                                                                                16:13:02
Traceback (most recent call last):
  File "/usr/lib/python3.13/site-packages/gnuradio/grc/gui_qt/components/canvas/block.py", line 408, in itemChange
    return QGraphicsItem.itemChange(self, change, value)
           ~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^
TypeError: unable to convert a QVariant back to a Python object
[1]    1117988 IOT instruction (core dumped)  gnuradio-companion --qt
```

haakov, 2025-02-05:
Thank you for the report! Are you using Pyside?

What does it say in the console subwindow when you start GRC Qt?

bczhc, 2025-02-06:
Hello. I'm not familiar with Python and how can I check if GRC is using pyside? Arch repository has pyside, but they're not installed locally on my machine. However, I do have pyside6 under this directory `/usr/lib/python3.13/site-packages`

```console
~ ❯ pacman -Q pyside2 pyside6                                                                 11:57:08
pyside2 5.15.13-1
pyside6 6.8.1.1-2
~ ❯ pacman -Qqe | rg -i pyside                                                                11:57:16
~ ❯                                                                                     ✘ 0|1 11:57:22
~ ❯ du -sh /usr/lib/python3.13/site-packages/PySide6                                          12:01:35
54M     /usr/lib/python3.13/site-packages/PySide6
```

GRC console just says:
```
[I] | <<< Welcome to GNU Radio Companion 3.10.11.0 >>>

  PyQt 5.15.11
  GUI preferences file: /home/bczhc/.config/gnuradio/grc_qt.conf
  Block paths:
 	/home/bczhc/.local/state/gnuradio
 	/usr/share/gnuradio/grc/blocks
  Using hicolor icon theme
```
When deleting a block, GRC just immediately crashes, with the traceback printed on the terminal console.

haakov, 2025-02-06:
Okay, so you do have Pyside6 installed, but you're using PyQt5 (printed in the console). 

I'm not able to reproduce the crash with PyQt6, but I'll try with PyQt5.

I'd be interested to know what happens if you run
`export QT_API="pyqt6"` before you run GRC (it needs to be in the same terminal). You might have to install the python-pyqt6 package first.

bczhc, 2025-02-06:
With `QT_API=pyqt6` it works now! No crash. Thanks. So I've been having `python-pyqt6` installed already, and GRC is still using pyqt5? Can this be configured or I need to run it only with this env?

haakov, 2025-02-08:
I've added a PR: https://github.com/gnuradio/gnuradio/pull/7723