gnuradio/gnuradio #7716: Is it time to retire CMakeParseArgumentsCopy?

haakov, 2025-02-04:
### What happened?

While working on https://github.com/gnuradio/gnuradio/issues/7708, I was having trouble getting past the configure step due to this error:
```
CMake Error at /usr/lib/cmake/Qt6/QtPublicFindPackageHelpers.cmake:34 (message):
  PACKAGE_NAME is required
Call Stack (most recent call first):
  /usr/lib/cmake/Qt6/QtPublicDependencyHelpers.cmake:42 (_qt_internal_get_package_components_id)
  /usr/lib/cmake/Qt6/Qt6Dependencies.cmake:34 (_qt_internal_find_third_party_dependencies)
  /usr/lib/cmake/Qt6/Qt6Config.cmake:155 (include)
  gr-qtgui/CMakeLists.txt:16 (find_package)
```

After some frustrating CMake debugging, I realized that GR has its own (copied from CMake) implementation of `cmake_parse_arguments()`:
https://github.com/gnuradio/gnuradio/blob/main/cmake/Modules/CMakeParseArgumentsCopy.cmake
This is included here:
https://github.com/gnuradio/gnuradio/blob/58b3b649b69e0248a08b77e7f49ef9f2c45ac333/cmake/Modules/GrPython.cmake#L204
which means that the GR version is used in all subsequent calls to `cmake_parse_arguments()`, instead of the CMake-native version. I tried removing the include call, and suddenly everything worked as I'd expected. 

**The question is, do we still need this file?** 

### System Information

OS: Arch Linux
GR Installation Method: Source
CMake 3.31.5
Qt 6.8.2
qwt (Qt6) 6.3.0

### GNU Radio Version

3.11-git (main)

### Steps to Reproduce the Problem

Check out https://github.com/haakov/gnuradio/tree/qt5to6 and try configuring with `cmake .. -DTRY_QT6_BUILD=1`

(if it works somehow, you might be using the Qt5 version of Qwt) 

### Relevant log output

```shell

```

willcode, 2025-02-04:
This file is prehistoric (at least wrt Git history). I'm guessing this was copied in back when it was an optional cmake module.

```
#[=======================================================================[.rst:
CMakeParseArguments
-------------------

This module once implemented the :command:`cmake_parse_arguments` command
that is now implemented natively by CMake.  It is now an empty placeholder
for compatibility with projects that include it to get the command from
CMake 3.4 and lower.
#]=======================================================================]
```

@marcusmueller I think it can/should go if the native version works.

Kangie, 2025-02-05:
Yeah, get rid of it. The minimum CMake version supported in the project is 3.16.