gnuradio/gnuradio #7724: QSocketNotifier: Can only be used with threads started with QThread

Pauloux, 2025-02-08:
### What happened?

Hello, I installed GNUradio on Linux (Ubuntu 24.04.1 LTS) using `sudo apt-get install gnuradio` as recommended on the GNUradio website. However, when I run a simple script (Signal Source, Throttle, QT GUI Time Sink), the graphs produced do not move and the following message appears: `QSocketNotifier: Can only be used with threads started with QThread`.
I completely reinstalled Ubuntu from scratch because I had an old distribution updated several times and I thought I could be the source of my problems but the problems persist.
I would like to use GNUradio for my university program and for personal uses after because I am interested in SDR and open source software in general.

I apologize if this is not considered a bug bug I could not find any information about this error related to GNUradio on the Internet. I hope my report will help to improve GNUradio and other users with the same issue.

Pauloux

### System Information

OS: Ubuntu 24.04.1 LTS)
GR Installation Method: `sudo apt-get install gnuradio`

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.9.2

### Steps to Reproduce the Problem

- Create a file and make a flow graph like the following : 
[image]
- Execute the script

The result I get (this graph stays still and do not update as it should): 
[image]

### Relevant log output

```shell
<<< Welcome to GNU Radio Companion 3.10.9.2 >>>

Block paths:
	/usr/share/gnuradio/grc/blocks

Loading: "/home/pauloux/Bureau/untitled.grc"
>>> Done

Generating: "/home/pauloux/Bureau/untitled.py"

Executing: /usr/bin/python3 -u /home/pauloux/Bureau/untitled.py

QSocketNotifier: Can only be used with threads started with QThread

>>> Done
```

Pauloux, 2025-02-20:
Is there someone on this forum ?

willcode, 2025-02-20:
@Pauloux this is not a forum, it's an issue tracker for bug reports and feature request. Since the time sink in the screenshot above is set for 1M points, I'm going to say this is not a bug. Please use chat.gnuradio.org (General Chat room) or the mailing list for help requests.

closed 2025-02-20 by willcode