gnuradio/gnuradio #7725: [Port older fix to v3.10] UHD: USRP source block does not generate stream tags when RX frequency changed via its message port.

jcfitzpatrick12, 2025-02-08:
### What happened?

_Context_
I am developing a program called [`spectre`](https://github.com/jcfitzpatrick12/spectre), which is a receiver agnostic program used to record radio spectrograms. Currently, I am working to provide support for the [b200-mini](https://www.ettus.com/all-products/usrp-b200mini/) receiver via the USRP source block. 

_Problem_
I use message passing to re-tune the center frequency of SDRs, and require that source blocks produce a stream tag indicating at which sample the frequency re-tuning occurred. Generally, if the USRP source block's centre frequency is altered then the output stream is tagged with the current frequency. However, I am experiencing an issue where if the center frequency is altered via message passing, then these tags are not generated. 

_Proposed Resolution_
In trying to resolve this, I noticed it has been previously reported and solved in [this issue](https://github.com/gnuradio/gnuradio/issues/1357). However, the fix was implemented (see [this commit](https://github.com/gnuradio/gnuradio/commit/61d199470e6f350e993e7d7953914a91d6a8b4b7)) in GNU Radio [v3.8.5.0](https://github.com/gnuradio/gnuradio/releases/tag/v3.8.5.0). It does not look like it has yet been ported to 3.10, after inspecting the corresponding source code in later versions: https://github.com/gnuradio/gnuradio/blob/v3.10.1.1/gr-uhd/lib/usrp_source_impl.cc#L114-L124. 

Probably, the root cause is identical to the previously reported issue. Quoting from there: 

"The root cause is that command messages use the SET_CENTER_FREQ_FROM_INTERNALS macro to set the RX frequency instead of the set_center_freq() function. The macro does not set the _tag_now flag, so no tags are generated."

And the fix would be expected to be the same, I think. 

### System Information

OS: (Ubuntu 22.04 base image, see the [Dockerfile](https://github.com/jcfitzpatrick12/spectre/blob/main/backend/Dockerfile))
GR Installation Method: (native distro packages - apt)

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.1.1

### Steps to Reproduce the Problem

To reproduce my environment with Docker, you can follow the "Quick Start for Developers" section in the [README](https://github.com/jcfitzpatrick12/spectre/blob/main/README.md) for `spectre`. I'm currently working on the branch "b200-mini-support". However, the problem should be reproducible in any environment as long as it is running GNU Radio v3.10.

- Create a USRP source block
- Re tune it via message passing.
- Connect the output of the USRP source to a tag debug, to show the absence of tags.

Today, I'll attach a flow graph to demonstrate this.

### Relevant log output

```shell

```

willcode, 2025-02-08:
@mbr0wn should the channel map be used here, or is that already resolved at this point and `chan` should be used as suggested?

jcfitzpatrick12, 2025-02-08:
@willcode I attached a flow-graph which can be used to reproduce the results, along with the corresponding logs. There is an initial tag generated when the stream starts, but no further tags on manually sending messages via the QT GUI Message Edit block. On inspecting the waterfall plot, it can be seen the frequency is apparently updating as expected, just no tags along with it.

jcfitzpatrick12, 2025-02-12:
@willcode I am unsure of the usual process, but I am happy to fork, build from source for 3.10 and try the fix in practice. If it works, I can raise a pull request. Though, I am quite unfamiliar with the code - and what impacts it might have elsewhere ! Happy to do whatever I can on my end to try and resolve this - just let me know, or point me to any relevant documentation :)

willcode, 2025-02-12:
I'd like to get agreement from @mbr0wn or @gmeyerho before you go through the trouble.

gmeyerho, 2025-02-13:
@willcode @jcfitzpatrick12 @mbr0wn - I have not worked much with the USRP source block, but looking at the solution in 3.8, I don't see any issue with that being ported to 3.10 as well. It looks like functionally the only change is that _tag_now is set to true, which seems like it will have the desired behavioral change.

gmeyerho, 2025-02-17:
@jcfitzpatrick12 - I generally will fork and test with the main branch, and target the PR at the gnuradio:main branch with the tags gr-uhd and Backport-3.10

jcfitzpatrick12, 2025-02-17:
@gmeyerho I'll do exactly that, and report back :) 

One more very rough (probably ill-posed) curiosity-based question, what is the timescale when a PR is approved, and being able to apt-get install gnuradio with the change included?

willcode, 2025-02-17:
That's hard to know. We're putting out a new release now, and that will probably be the one picked up by the distros. We have not updated the standalone PPA recently, so the only apt source is via distros, subject to their schedules.

jcfitzpatrick12, 2025-02-21:
Apologies, accidentally closed this somehow. Thank you for the information on the release process ! I’ll report back when I’ve tested the fix.

jcfitzpatrick12, 2025-02-25:
I have implemented the fix as we discussed, tested it and [raised a PR](https://github.com/gnuradio/gnuradio/pull/775). Although, I wasn't able to add a tag for "Backport-3.10". Let me know what the actions from here, and if there are any necessary amendments to make.

Using the fix, I'm now able to make some nice swept spectrograms with `spectre` with any USRP device. Here's one hot-of-the-press over the FM band using my b200mini and a rubber ducky antenna:  

[image]

marcusmueller, 2025-02-27:
@jcfitzpatrick12 reason it wasn't tagged as to be backported (or being discussed as not backportable) is because we're [in the middle of a maintainer switchover](https://www.gnuradio.org/news/2025-02-20-saying-thank-you-to-jeff/), and the subsystem maintainer (@mbr0wn) or someone else trusted hasn't gotten around to categorizing your PR yet :)

I don't have the time right now to review this, and I think it'd be really better when someone with more in-depth knowledge of this PR family does that; so I'll simply have to ask you for a little more patience :)

jcfitzpatrick12, 2025-02-27:
@marcusmueller Thank you for the update, completely understandable. As before, on standby to help :)

PR #7751 (2025-03-20): uhd: Generate tags when frequency changed from message port.

## Description
Generally, if the USRP source block's centre frequency is altered, then the output stream is tagged with the current frequency. However, at present if the center frequency is altered via message passing, then these tags are not generated. This PR fixes this functionality by using the `set_center_freq` member function. This performs the same functionality as the original code with the addition of properly setting the `_tag_now` flag.

## Related Issue
Fixes #7725 

## Which blocks/areas does this affect?

The "USRP Sou [... cut]
closed 2025-03-20 by mbr0wn