gnuradio/gnuradio #7731: C++ Generation wrong: QCloseEvent

gcarmix, 2025-02-15:
### What happened?

When gnuradio-companion generates a C++ code, in No Gui mode, I noticed that the code always generate multiple instances of this method:

top::closeEvent(QCloseEvent *event) {
this->settings->setValue("geometry",this->saveGeometry());
event->accept();
}

that is no needed at all, since I'm not even using QT

### System Information

OS: Ubuntu 22.04
GR Installation Method: Source

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.11.0

### Steps to Reproduce the Problem

simply create a project, disable the qt gui, set the project to C++

### Relevant log output

```shell

```

ZakyHermawan, 2025-02-19:
This bug is fixed in https://github.com/gnuradio/gnuradio/commit/ad86524dbf46c90ce50ab0951b81f2b5ccff7290

the PR is also getting backported to 3.10 [here](https://github.com/gnuradio/gnuradio/blob/maint-3.10/grc/core/generator/cpp_templates/flow_graph.cpp.mako)

ZakyHermawan, 2025-02-25:
@willcode, do you think this issue should be closed since it has already been fixed?

willcode, 2025-02-25:
Yes, this is the same problem as in #7500, which was marked as fixed.

closed 2025-02-25 by willcode