gnuradio/gnuradio #7752: early-late gate ted should have a computation depth of 3

n-west, 2025-02-28:
### What happened?

The `ted_early_late` initializes with `inputs_per_symbol` of 2 and a `error_computation_depth` of 2.
https://github.com/gnuradio/gnuradio/blob/2acc77ab20ff7e8574076f33a859daccdd4eb2a1/gr-digital/lib/timing_error_detector.h#L351

The `error_computation_depth` is only used to create a the buffer of ted inputs (https://github.com/gnuradio/gnuradio/blob/2acc77ab20ff7e8574076f33a859daccdd4eb2a1/gr-digital/lib/timing_error_detector.cc#L208)

The ted_early_late error_computation uses the gardner style (early - late)*prompt https://github.com/gnuradio/gnuradio/blob/2acc77ab20ff7e8574076f33a859daccdd4eb2a1/gr-digital/lib/timing_error_detector.cc#L299

That directly indexes the d_input buffer that is error_computation_depth long. So d_input has a size() of 2, but is indexed with 0, 1, 2 (i.e., 3 elements required). The buffer should be 1 item longer since this dereference will always be out of bounds.

The computation_depth should be 3 to make this buffer the correct length.

I also suspect that the inputs_per_symbol should be 3, one each for the early, late, prompt sample. This is slightly confusing because this is not "just" an early-late gate, it's a Gardner early-late gate where the error is scaled by signal amplitude. Dan Boschen has a detailed writeup here: https://dsp.stackexchange.com/a/61809.

There's another minor issue if comparing that SO writeup: the GR version is using prompt*(early - late) and the SO writeup suggests prompt*(late-early). Consulting my 2nd ed. copy of Sklar's Digital Communications text, p. 629 suggests the GR form of early-late is correct, but it likely just depends on how you use the error to adjust the resampler clock phase, so I suspect the GR order is just fine in that difference from the SO writeup.

### System Information

N/A. I only discovered this while reading code, and haven't observed bad behavior while running. I also haven't tested the ted_early_late to see if it works as is or not.

### GNU Radio Version

3.11-git (main)

### Steps to Reproduce the Problem

I haven't observed whether this is problematic in runtime or not

### Relevant log output

```shell

```

n-west, 2025-02-28:
I know @awalls-cx18 has done a lot of really good work around these blocks, so might be able to provide some insight to the samples per symbol issue

closed 2025-03-03 by marcusmueller

awalls-cx18, 2025-03-04:
> I know [@awalls-cx18](https://github.com/awalls-cx18) has done a lot of really good work around these blocks, so might be able to provide some insight to the samples per symbol issue.
> 
> I mixed myself and spun in circles on how the (late-error) * middle is affecting the lock. I removed that from the initial report because I think I was wrong. Sorry for the tag. I am still thinking through it, but also think it might be the case that the error should be the difference |early| - |late|, or the difference of absolute values.

An Early-Late TED is the small signal ML TED with the derivative term replaced with a central difference filtered set of inputs (the esrly-late part).  The "middle" term is the part of the ML TED expression that is left alone.

PR #7754 (2025-03-03): gr-digital: fix early_late ted computation depth to match actual depth
Fixes #7752

- [ X ] I have read the [CONTRIBUTING document](https://github.com/gnuradio/gnuradio/blob/main/CONTRIBUTING.md).
- [ X ] I have squashed my commits to have one significant change per commit. 
- [x] I [have signed my commits before making this PR](https://github.com/gnuradio/gnuradio/blob/main/CONTRIBUTING.md#dco-signed)
- [ X ] My code follows the code style of this project. See [GREP1.md](https://github.com/gnuradio/greps/blob/main/grep-0001-coding-guidelines.md).
- [ N/A ] I have updated [the documentation](h [... cut]