gnuradio/gnuradio #7756: OOT template uses MANIFEST.yml that rest of the ecosystem (CGRAN, GRC Qt) does not yet support

ryanvolz, 2025-03-03:
### What happened?

I've been rebuilding and updating some OOT modules for radioconda, and one thing I did was update the manifest to the new YAML format as seen in the OOT module template. I started to put in some PRs to have this change included upstream in individual OOT modules, but I noticed that:

1. CGRAN still only looks for `MANIFEST.md`
2. The GRC Qt OOT module browser still doesn't fully use the new YAML format and doesn't look for manifests in the directory to which the OOT module template installs them. (It looks within the `gui-qt/resources/manifests` directory where only an example lives and not the installed path).

I figure this is more of an in-process transition than something that has been completed (particularly with Qt GRC), but I didn't see that this was tracked anywhere yet.

Related to #7123.

### System Information

N/A

### GNU Radio Version

3.11-git (main)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

N/A

### Relevant log output

```shell

```

marcusmueller, 2025-03-03:
As this interfaces with CGRAN, I'd like to bring in @777arc

777arc, 2025-03-03:
So currently CGRAN grabs the top portion of the MANIFEST file (everything above ---) and does a `yaml.safe_load` then grabs the following fields if they exist:

```
processed_yaml.get('gr_supported_version', '')
processed_yaml.get('dependencies', ['None'])
tags = ", ".join(processed_yaml.get('tags',['None'])),
description = processed_yaml.get('brief', 'None'),
author = ", ".join(processed_yaml.get('author', ['None'])),
copyright_owner = ", ".join(processed_yaml.get('copyright_owner', ['None'])),
icon = validate_icon_URL(processed_yaml.get('icon', '')),
website = processed_yaml.get('website', 'None'),
```

But it's not like we could change any of that without also changing dozens of existing MANIFEST files

ryanvolz, 2025-03-03:
> But it's not like we could change any of that without also changing dozens of existing MANIFEST files

Sure, and I would expect many OOTs to never migrate to the new format. My suggestion would be to search first for `MANIFEST.yml` and use that if it exists, but then quietly fall back to `MANIFEST.md` and use the existing parsing otherwise.

777arc, 2025-03-03:
Ah yep that's easy to add, is there an OOT with the yaml version live on its main branch that I can use for testing?

ryanvolz, 2025-03-03:
> Ah yep that's easy to add, is there an OOT with the yaml version live on its main branch that I can use for testing?

Not yet AFAIK, but I have PRs in to [gr-rds](https://github.com/bastibl/gr-rds/pull/98) and [gr-inspector](https://github.com/gnuradio/gr-inspector/pull/57) that stand a chance of being merged soon (the latter someone like @marcusmueller could merge ;-) ).

marcusmueller, 2025-03-09:
FYI, merged gr-inspector PR

777arc, 2025-03-09:
alright I updated CGRAN to use the yml first, if it exists, CGRAN should update overnight https://github.com/gnuradio/cgran/commit/f6f04f1b5281734b886f9c96f2bbc3dd556d052b