gnuradio/gnuradio #7765: GnuRadio-Companion startup fails to initialize GTK

NetScr1be, 2025-03-10:
### What happened?

- not running over SSH
- 

> patrick@dell:~/radio/sdr$ /usr/bin/gnuradio-companion
> While trying to display an error message, another error occurred
> Namespace Gtk not available
> The original error message follows.
> ValueError

> Failed to initialize GTK. If you are running over ssh, did you enable X forwarding and start ssh with -X?

Noticed previous instances of this bug closed without resolution

### System Information

OS: Debian Linux

> Distributor ID:	Debian
> Description:	Debian GNU/Linux 12 (bookworm)
> Release:	12
> Codename:	bookworm

GR Installation Method: native distro packages

```
patrick@dell:~/radio/sdr$ sudo apt-get install gnuradio

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
gnuradio is already the newest version (3.10.5.1-3).
```

```
patrick@dell:~/radio/sdr$ gnuradio-config-info --print-all
/usr/
/etc
/etc/gnuradio/conf.d
/home/patrick/.gnuradio
Sun, 07 May 2023 14:09:50Z
testing-support;python-support;post-install;doxygen;man-pages;gnuradio-runtime;common-precompiled-headers;gr-ctrlport;* thrift;gnuradio-companion;JSON/YAML config blocks;gr-blocks;gr-fec;gr-fft;gr-filter;gr-analog;gr-digital;gr-dtv;gr-audio;* alsa;* oss;* jack;* portaudio;gr-channels;gr-pdu;gr-iio;* libad9361;gr-qtgui;gr-trellis;gr-uhd;gr-uhd UHD 4.0 RFNoC;gr-utils;gr_modtool;gr_blocktool;gr-video-sdl;gr-vocoder;* codec2;* freedv;* gsm;gr-wavelet;gr-zeromq;gr-network;gr-soapy
3.10.5.1
cc (Debian 12.2.0-14) 12.2.0 
Copyright (C) 2022 Free Software Foundation, Inc. 
This is free software see the source for copying conditions.  There is NO 
warranty not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
c++ (Debian 12.2.0-14) 12.2.0 
Copyright (C) 2022 Free Software Foundation, Inc. 
This is free software see the source for copying conditions.  There is NO 
warranty not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
/usr/bin/cc:::-O3 -DNDEBUG -g -O2 -ffile-prefix-map=/build/gnuradio-4HxWIe/gnuradio-3.10.5.1=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized 
/usr/bin/c++:::-O3 -DNDEBUG -g -O2 -ffile-prefix-map=/build/gnuradio-4HxWIe/gnuradio-3.10.5.1=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
2.10.3
```

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.5.1

### Steps to Reproduce the Problem

```
patrick@dell:~/radio/sdr$ gnuradio-companion
```

### Relevant log output

```shell
While trying to display an error message, another error occurred
Namespace Gtk not available
The original error message follows.
ValueError

Failed to initialize GTK. If you are running over ssh, did you enable X forwarding and start ssh with -X?

(Namespace Gtk not available)
```

NetScr1be, 2025-03-10:
There is no discussion of this on the mailing list (that I could find).

NetScr1be, 2025-03-10:
Your install package is missing componentry.

GRC works after doing;

```
sudo apt-get install libgtk-3-dev
```

Why force users to install a development package?

marcusmueller, 2025-03-10:
@NetScr1be hi! Thanks for the report.

> Why force users to install a development package?

Rest assured, that wasn't our intent. (and the fact that it seems to work for so many other debian users might indicate something else being wrong. Let me think about that.)

My best guess here is that somehow, `python3-gi` on your system got broken, somehow? (or something like `gir1.2-gtk`?) and installing `libgtk-3-dev` somehow fixed that. Strange, because debian is rather robust w.r.t. that.

Could you do me a favor? I'd like to know what dependencies got installed *alongside* `libgtk-3-dev` when you ran `sudo apt-get install libgtk-3-dev`. Could you 

```shell
grep -E " (install|remove) " /var/log/dpkg.log > /tmp/packagechanges.txt
```

and then use your favorite text editor to open /tmp/packagechanges.txt and find all the lines from the same change (i.e., things that got installed / removed alongside with libgtk-3-dev, so, a couple seconds around the line

```text
2025-03-10 01:xx:xx install libgtk-3-dev:amd64 <none> 3.24.38-2~deb12u3
```
or such.

Save these lines to a .txt file and use the "Paste drop or click to add files" below the comment field here to attach  (github is annoyingly cranky about file extensions).

NetScr1be, 2025-03-10:
@marcusmueller thanks for the prompt response. As someone who has spent the past 20 years supporting PoS, fiat card and crypto payment solutions I understand the difficulty making things work on the myriad platforms that exist.

The searches I did on the topic all talked about libgtk-3-dev so, as much as I hate to shotgun solutions, I thought I was worth the time and disk space to try it. Was kind of surprised it worked.

I have gone on to successfully complete a couple of the flowgraph tutorials making a broadcast FM tuner with the new RTL-SDR dongle that triggered the need for GRC as well as adapting the [spectrum analyzer tutorial](https://wiki.gnuradio.org/index.php?title=Guided_Tutorial_Hardware_Considerations) to work with the RTL-SDR by replacing the UHD USRP Source block with the Soapy RTLSDR source and guessing at the required parameters. It crashes if the frequency is changed too far/fast/both (haven't figured that out (wondering about the step parameter)) but it does start up and show something even if I haven't gotten it to show a signal yet.

The whole goal of this was to use the RTL-SDR dongle to do realtime ADSB signal capture and processing in conjunction with commercial drone operations. All this as a way of adding value and safety to commercial drone flights.

As per your request for more info;

```
patrick@dell:~/radio/sdr$ sudo grep -E " (install|remove) " /var/log/dpkg.log > /tmp/packagechanges.txt

patrick@dell:~/radio/sdr$ ls -al /tmp/packagechanges.txt 
-rw-r--r-- 1 patrick patrick 94959 Mar 10 12:11 /tmp/packagechanges.txt
```

[image]

Copy **ALL** log messages from relevant time period and paste them here.

N.B. first line is *not* relevant but included to show nothing was missed

```
2025-03-05 16:43:01 install xtrx-dkms:all <none> 0.0.1+git20190320.5ae3a3e-3.2
2025-03-09 20:26:20 install gir1.2-atk-1.0:amd64 <none> 2.46.0-5
2025-03-09 20:26:21 install gir1.2-freedesktop:amd64 <none> 1.74.0-3
2025-03-09 20:26:21 install gir1.2-atspi-2.0:amd64 <none> 2.46.0-5
2025-03-09 20:26:22 install gir1.2-gdkpixbuf-2.0:amd64 <none> 2.42.10+dfsg-1+deb12u1
2025-03-09 20:26:22 install gir1.2-harfbuzz-0.0:amd64 <none> 6.0.0+dfsg-3
2025-03-09 20:26:23 install libpangoxft-1.0-0:amd64 <none> 1.50.12+ds-1
2025-03-09 20:26:24 install gir1.2-pango-1.0:amd64 <none> 1.50.12+ds-1
2025-03-09 20:26:24 install gir1.2-gtk-3.0:amd64 <none> 3.24.38-2~deb12u3
2025-03-09 20:26:25 install libffi-dev:amd64 <none> 3.4.4-1
2025-03-09 20:26:26 install libglib2.0-bin:amd64 <none> 2.74.6-2+deb12u5
2025-03-09 20:26:27 install libglib2.0-dev-bin:amd64 <none> 2.74.6-2+deb12u5
2025-03-09 20:26:28 install uuid-dev:amd64 <none> 2.38.1-5+deb12u3
2025-03-09 20:26:28 install libblkid-dev:amd64 <none> 2.38.1-5+deb12u3
2025-03-09 20:26:29 install libsepol-dev:amd64 <none> 3.4-2.1
2025-03-09 20:26:30 install libpcre2-32-0:amd64 <none> 10.42-1
2025-03-09 20:26:30 install libpcre2-posix3:amd64 <none> 10.42-1
2025-03-09 20:26:31 install libpcre2-dev:amd64 <none> 10.42-1
2025-03-09 20:26:32 install libselinux1-dev:amd64 <none> 3.4-1+b6
2025-03-09 20:26:33 install libmount-dev:amd64 <none> 2.38.1-5+deb12u3
2025-03-09 20:26:34 install libglib2.0-dev:amd64 <none> 2.74.6-2+deb12u5
2025-03-09 20:26:35 install libatk1.0-dev:amd64 <none> 2.46.0-5
2025-03-09 20:26:35 install libdbus-1-dev:amd64 <none> 1.14.10-1~deb12u1
2025-03-09 20:26:36 install libxfixes-dev:amd64 <none> 1:6.0.0-2
2025-03-09 20:26:37 install libxi-dev:amd64 <none> 2:1.8-1+b1
2025-03-09 20:26:38 install libxtst-dev:amd64 <none> 2:1.2.3-1.1
2025-03-09 20:26:38 install libatspi2.0-dev:amd64 <none> 2.46.0-5
2025-03-09 20:26:39 install libatk-bridge2.0-dev:amd64 <none> 2.46.0-5
2025-03-09 20:26:40 install libbrotli-dev:amd64 <none> 1.0.9-2+b6
2025-03-09 20:26:40 install liblzo2-2:amd64 <none> 2.10-2
2025-03-09 20:26:41 install libcairo-script-interpreter2:amd64 <none> 1.16.0-7
2025-03-09 20:26:42 install libpng-dev:amd64 <none> 1.6.39-2
2025-03-09 20:26:42 install libfreetype-dev:amd64 <none> 2.12.1+dfsg-5+deb12u3
2025-03-09 20:26:43 install libfontconfig-dev:amd64 <none> 2.14.1-4
2025-03-09 20:26:44 install libpixman-1-dev:amd64 <none> 0.42.2-1
2025-03-09 20:26:44 install libxcb-render0-dev:amd64 <none> 1.15-1
2025-03-09 20:26:45 install libxcb-shm0-dev:amd64 <none> 1.15-1
2025-03-09 20:26:46 install libxrender-dev:amd64 <none> 1:0.9.10-1.1
2025-03-09 20:26:46 install libcairo2-dev:amd64 <none> 1.16.0-7
2025-03-09 20:26:47 install libdatrie-dev:amd64 <none> 0.2.13-2+b1
2025-03-09 20:26:48 install libdeflate-dev:amd64 <none> 1.14-1
2025-03-09 20:26:49 install libegl1-mesa-dev:amd64 <none> 22.3.6-1+deb12u1
2025-03-09 20:26:49 install libepoxy-dev:amd64 <none> 1.5.10-1
2025-03-09 20:26:50 install libfribidi-dev:amd64 <none> 1.0.8-2.1
2025-03-09 20:26:51 install libjpeg62-turbo-dev:amd64 <none> 1:2.1.5-2
2025-03-09 20:26:51 install libjpeg-dev:amd64 <none> 1:2.1.5-2
[... 4 lines cut ...]
2025-03-09 20:26:54 install libtiffxx6:amd64 <none> 4.5.0-6+deb12u2
2025-03-09 20:26:55 install liblerc-dev:amd64 <none> 4.0.0+ds-2
2025-03-09 20:26:56 install libtiff-dev:amd64 <none> 4.5.0-6+deb12u2
2025-03-09 20:26:56 install libgdk-pixbuf-2.0-dev:amd64 <none> 2.42.10+dfsg-1+deb12u1
2025-03-09 20:26:57 install libgraphite2-dev:amd64 <none> 1.3.14-1
2025-03-09 20:26:58 install libharfbuzz-icu0:amd64 <none> 6.0.0+dfsg-3
2025-03-09 20:26:58 install libharfbuzz-gobject0:amd64 <none> 6.0.0+dfsg-3
2025-03-09 20:26:59 install libharfbuzz-subset0:amd64 <none> 6.0.0+dfsg-3
[... cut ...]
2025-03-09 20:27:05 install libxcursor-dev:amd64 <none> 1:1.2.1-1
2025-03-09 20:27:05 install libxdamage-dev:amd64 <none> 1:1.1.6-1
2025-03-09 20:27:06 install libxinerama-dev:amd64 <none> 2:1.1.4-3
2025-03-09 20:27:07 install libxkbcommon-dev:amd64 <none> 1.5.0-1
2025-03-09 20:27:07 install libxrandr-dev:amd64 <none> 2:1.5.2-2+b1
2025-03-09 20:27:08 install wayland-protocols:all <none> 1.31-1
2025-03-09 20:27:09 install libgtk-3-dev:amd64 <none> 3.24.38-2~deb12u3
2025-03-09 20:27:10 install libpng-tools:amd64 <none> 1.6.39-2
```

marcusmueller, 2025-03-10:
probable culprit:

`2025-03-09 20:26:24 install gir1.2-gtk-3.0:amd64 <none> 3.24.38-2~deb12u3`

that should have come with the python-gtk dependency that GNU Radio has. Need to investigate this. This might end up being a debian bug. 

I might close this issue here soon, because while, as project, on good terms with our debian packager, we're not debian, and can't do anything about the packaging. But I'd like to check back with him before closing this.

NetScr1be, 2025-03-10:
Please keep me in the loop on your discussions with them.

If it comes out as an issue with their packaging I will follow it up through that channel.

Thanks again for you prompt attention.

I can't take up the issue with the RTL-SDR spectrum analyzer on the GRC discussion email list because they haven't confimed ny subscription. Do you happen to have any contact with them internally? Can you point me at who I need to prod to move this forward?

Does Gnu need support talent? Who might be able to answer that question?

Thanks again,

--PB--

NetScr1be, 2025-03-10:
Remove libgtk-3-dev as per previous @marcusmueller comment;

```
patrick@dell:~/radio/sdr$ sudo apt-get remove --autoremove -x libgtk-3-dev

E: Command line option 'x' [from -x] is not understood in combination with the other options.
```

There is no -x option on the [apt-get man page](https://linux.die.net/man/8/apt-get).

NetScr1be, 2025-03-10:
Remove libgtk-3-dev (without '-x' option)

```
patrick@dell:~/radio/sdr$ sudo apt-get remove --autoremove libgtk-3-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages will be REMOVED:
  gir1.2-atspi-2.0 libatk-bridge2.0-dev libatk1.0-dev libatspi2.0-dev libblkid-dev libbrotli-dev libcairo-script-interpreter2
  libcairo2-dev libdatrie-dev libdbus-1-dev libdeflate-dev libegl1-mesa-dev libepoxy-dev libffi-dev libfontconfig-dev
  libfreetype-dev libfribidi-dev libgdk-pixbuf-2.0-dev libglib2.0-bin libglib2.0-dev libglib2.0-dev-bin libgraphite2-dev
  libgtk-3-dev libharfbuzz-dev libharfbuzz-gobject0 libharfbuzz-icu0 libharfbuzz-subset0 libjbig-dev libjpeg-dev libjpeg62-turbo-dev
  liblerc-dev liblzma-dev liblzo2-2 libmount-dev libpango1.0-dev libpcre2-32-0 libpcre2-dev libpcre2-posix3 libpixman-1-dev
  libpng-dev libpng-tools libselinux1-dev libsepol-dev libthai-dev libtiff-dev libtiffxx6 libwayland-bin libwayland-dev libwebp-dev
  libxcb-render0-dev libxcb-shm0-dev libxcomposite-dev libxcursor-dev libxdamage-dev libxfixes-dev libxft-dev libxi-dev
  libxinerama-dev libxkbcommon-dev libxrandr-dev libxrender-dev libxtst-dev libzstd-dev pango1.0-tools uuid-dev wayland-protocols
0 upgraded, 0 newly installed, 66 to remove and 0 not upgraded.
After this operation, 70.2 MB disk space will be freed.
Do you want to continue? [Y/n] 
(Reading database ... 136866 files and directories currently installed.)
Removing libgtk-3-dev:amd64 (3.24.38-2~deb12u3) ...
Removing libatk-bridge2.0-dev:amd64 (2.46.0-5) ...
Removing libatspi2.0-dev:amd64 (2.46.0-5) ...
**Removing gir1.2-atspi-2.0:amd64 (2.46.0-5) ...**
Removing libatk1.0-dev:amd64 (2.46.0-5) ...
Removing libpango1.0-dev:amd64 (1.50.12+ds-1) ...
Removing libharfbuzz-dev:amd64 (6.0.0+dfsg-3) ...
Removing libxft-dev:amd64 (2.3.6-1) ...
Removing libcairo2-dev:amd64 (1.16.0-7) ...
Removing libfontconfig-dev:amd64 (2.14.1-4) ...
Removing libfreetype-dev:amd64 (2.12.1+dfsg-5+deb12u3) ...
Removing libbrotli-dev:amd64 (1.0.9-2+b6) ...
Removing libcairo-script-interpreter2:amd64 (1.16.0-7) ...
Removing libthai-dev:amd64 (0.1.29-1) ...
Removing libdatrie-dev:amd64 (0.2.13-2+b1) ...
Removing libdbus-1-dev:amd64 (1.14.10-1~deb12u1) ...
Removing libgdk-pixbuf-2.0-dev:amd64 (2.42.10+dfsg-1+deb12u1) ...
Removing libtiff-dev:amd64 (4.5.0-6+deb12u2) ...
Removing libdeflate-dev:amd64 (1.14-1) ...
Removing libegl1-mesa-dev:amd64 (22.3.6-1+deb12u1) ...
Removing libepoxy-dev:amd64 (1.5.10-1) ...
Removing libwayland-dev:amd64 (1.21.0-1) ...
Removing libfribidi-dev:amd64 (1.0.8-2.1) ...
Removing libgraphite2-dev:amd64 (1.3.14-1) ...
Removing libharfbuzz-gobject0:amd64 (6.0.0+dfsg-3) ...
Removing libharfbuzz-icu0:amd64 (6.0.0+dfsg-3) ...
Removing libharfbuzz-subset0:amd64 (6.0.0+dfsg-3) ...
Removing libjbig-dev:amd64 (2.1-6.1) ...
Removing libjpeg-dev:amd64 (1:2.1.5-2) ...
Removing libjpeg62-turbo-dev:amd64 (1:2.1.5-2) ...
Removing liblerc-dev:amd64 (4.0.0+ds-2) ...
Removing liblzma-dev:amd64 (5.4.1-0.2) ...
Removing liblzo2-2:amd64 (2.10-2) ...
Removing libpixman-1-dev:amd64 (0.42.2-1) ...
Removing libpng-dev:amd64 (1.6.39-2) ...
Removing libpng-tools (1.6.39-2) ...
Removing libtiffxx6:amd64 (4.5.0-6+deb12u2) ...
Removing libwayland-bin (1.21.0-1) ...
Removing libwebp-dev:amd64 (1.2.4-0.2+deb12u1) ...
Removing libxcb-render0-dev:amd64 (1.15-1) ...
Removing libxcb-shm0-dev:amd64 (1.15-1) ...
Removing libxcomposite-dev:amd64 (1:0.4.5-1) ...
Removing libxcursor-dev:amd64 (1:1.2.1-1) ...
Removing libxdamage-dev:amd64 (1:1.1.6-1) ...
Removing libxtst-dev:amd64 (2:1.2.3-1.1) ...
Removing libxi-dev:amd64 (2:1.8-1+b1) ...
Removing libxfixes-dev:amd64 (1:6.0.0-2) ...
Removing libxinerama-dev:amd64 (2:1.1.4-3) ...
Removing libxkbcommon-dev:amd64 (1.5.0-1) ...
Removing libxrandr-dev:amd64 (2:1.5.2-2+b1) ...
Removing libxrender-dev:amd64 (1:0.9.10-1.1) ...
Removing libzstd-dev:amd64 (1.5.4+dfsg2-5) ...
Removing pango1.0-tools (1.50.12+ds-1) ...
Removing wayland-protocols (1.31-1) ...
Removing libglib2.0-dev:amd64 (2.74.6-2+deb12u5) ...
Removing libmount-dev:amd64 (2.38.1-5+deb12u3) ...
Removing libblkid-dev:amd64 (2.38.1-5+deb12u3) ...
Removing libffi-dev:amd64 (3.4.4-1) ...
Removing libglib2.0-bin (2.74.6-2+deb12u5) ...
Removing libglib2.0-dev-bin (2.74.6-2+deb12u5) ...
Removing libselinux1-dev:amd64 (3.4-1+b6) ...
Removing libpcre2-dev:amd64 (10.42-1) ...
Removing libpcre2-32-0:amd64 (10.42-1) ...
Removing libpcre2-posix3:amd64 (10.42-1) ...
Removing libsepol-dev:amd64 (3.4-2.1) ...
Removing uuid-dev:amd64 (2.38.1-5+deb12u3) ...
Processing triggers for libc-bin (2.36-9+deb12u9) ...
Processing triggers for man-db (2.11.2-2) ...
Processing triggers for sgml-base (1.31) ...
Processing triggers for libglib2.0-0:amd64 (2.74.6-2+deb12u5) ...
```
***GRC and broadcast FM flowgraph still works (!!??)***

Cheap shortcut to see if gir1.2-gtk-3.0 is still installed

```
patrick@dell:~/radio/sdr$ sudo apt-get install gir1.2-gtk-3.0
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
gir1.2-gtk-3.0 is already the newest version (3.24.38-2~deb12u3).
gir1.2-gtk-3.0 set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

SMH

NetScr1be, 2025-03-10:
If it matters, GRC functions after power-cycling the machine.

[image]

marcusmueller, 2025-03-10:
yeah, I think since autoremoving unused packages after removing libgtk-3-dev did not remove the gir12-gtk-3.0 package that our logs show was installed as dependency of libgtk-3-dev, I'm afraid this has to be some strangeness with the package cache on your specific machine, not really something we or debian's packaging can address!

closed 2025-03-10 by marcusmueller