gnuradio/gnuradio #7766: Fails to build on Fedora 43 (rawhide) with cmake 4.0 on s390x, ppc64le, it doesn't detect lib64

yarda, 2025-03-10:
### What happened?

The cmake-4.0 dropped `-DLIB_SUFFIX` variable initialization on fedora (cmake bug: https://bugzilla.redhat.com/show_bug.cgi?id=2351130). I don't know whether it was intentional or bug, but it reveals bug in gnuradio lib64 detection. The `lib64` is correctly detected on `x86_64`, but incorrectly detected now as a `lib` on ppc64le and s390x.

### System Information

OS: Fedora 43 (rawhide)
GR Installation Method: distro packaging from sources

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.12.0

### Steps to Reproduce the Problem

Build from sources. E.g. rebuild Fedora 42 SRPM for Fedora 43 on s390x or ppc64le.

Build logs from the builder:
https://koji.fedoraproject.org/koji/taskinfo?taskID=130091207

### Relevant log output

```shell
-- Installing: /builddir/build/BUILD/gnuradio-3.10.12.0-build/BUILDROOT/usr/lib/libgnuradio-pmt.so.3.10.12.0
-- Installing: /builddir/build/BUILD/gnuradio-3.10.12.0-build/BUILDROOT/usr/lib/libgnuradio-pmt.so.3.10.12
-- Installing: /builddir/build/BUILD/gnuradio-3.10.12.0-build/BUILDROOT/usr/lib/libgnuradio-pmt.so
-- Installing: /builddir/build/BUILD/gnuradio-3.10.12.0-build/BUILDROOT/usr/lib/cmake/gnuradio/gnuradio-pmtTargets.cmake
-- Installing: /builddir/build/BUILD/gnuradio-3.10.12.0-build/BUILDROOT/usr/lib/cmake/gnuradio/gnuradio-pmtTargets-release.cmake
-- Installing: /builddir/build/BUILD/gnuradio-3.10.12.0-build/BUILDROOT/usr/lib/cmake/gnuradio/gnuradio-pmtConfig.cmake
-- Installing: /builddir/build/BUILD/gnuradio-3.10.12.0-build/BUILDROOT/usr/lib/libgnuradio-runtime.so.3.10.12.0
-- Installing: /builddir/build/BUILD/gnuradio-3.10.12.0-build/BUILDROOT/usr/lib/libgnuradio-runtime.so.3.10.12
...
RPM build errors:
    File not found: /builddir/build/BUILD/gnuradio-3.10.12.0-build/BUILDROOT/usr/lib64/lib*.so.*
```

yarda, 2025-03-10:
I will go with the explict `-DLIB_SUFFIX` workaround on f43 for now.

LecrisUT, 2025-03-11:
I've opened https://github.com/gnuradio/gnuradio/pull/7768, do you want to try and pull it in and see if it covers everything?

yarda, 2025-03-11:
Thanks, I will check it.

PR #7768 (2025-07-22): cmake: Switch to standard `GNUInstallDirs`

## Description

`GNUInstallDirs` has been a standard since at least CMake 3.0 and distros are configuring it to their standard. It is better to use these as defaults instead.

## Related Issue
Fixes #7766

## Which blocks/areas does this affect?

## Testing Done