gnuradio/gnuradio #7769: Check with FreeBSD, still on 3.8.4.0

marcusmueller, 2025-03-11:
### Feature Description

FreeBSD was very fast adopting GNU Radio 3.8. It seems they're still on 3.8.4.0: https://www.freshports.org/comms/gnuradio

Maybe there's minor things, maybe a major bug that stops them from updating, maybe there's no demand, maybe I'm looking in the wrong place. Anyways, are they OK?

pinging @englishm on that

### Feature Urgency

low (just an idea)

englishm, 2025-03-11:
Good question! I haven't been actively using FreeBSD for ham radio in a while. I'd check with @DianeBruce. Could just be some dependencies are annoying to shift and there isn't much demand for newer versions yet.

marcusmueller, 2025-03-25:
Hi @DianeBruce! I'll ping you via the Fediverse :)