gnuradio/gnuradio #7771: Incorrect handling of Defines when generating bindings for OOT modules

PeterQFR, 2025-03-19:
### What happened?

when compiling an OOT module, the system errored with:
```
In file included from <built-in>:459:
<command line>:3:9: error: macro name must be an identifier
    3 | #define -U__PIE__ 1
      |         ^
1 error generated.

```
this output is from an undefine being considered a define in castxml command line. 

the verbose output from make showing this being called from 

```
/usr/bin/python3 /home/peter/software_defined_radio/gr-nacl/python/nacl/bindings/bind_oot_file.py --output_dir /home/peter/software_defined_radio/gr-nacl/python/nacl/bindings/.. --prefix /usr/local --module nacl --filename /home/peter/software_defined_radio/gr-nacl/python/nacl/bindings/../../../include/gnuradio/nacl/crypt_tagged_stream.h --status /home/peter/software_defined_radio/gr-nacl/build/python/nacl/bindings/crypt_tagged_stream_python.cc.regen_status --flag_automatic True --flag_pygccxml False
```

output from my defines being passed ot bind_oot_file was, i had cancelled out any defines:

defines ('',), 

includes /home/peter/software_defined_radio/gr-nacl/include/gnuradio/nacl,/home/peter/software_defined_radio/gr-nacl/include,/usr/local/include

I was able to trace it back to:
https://github.com/gnuradio/gnuradio/blob/4722ef5b2894be5bc69ddf669232be6eba1b0cca/gr-utils/blocktool/core/parseheader_generic.py#L316

Even though `self.define_symbols` was empty, if passed, pygccxml would generate a -D as part of the castxml command line. 
This error goes away when I change 

https://github.com/gnuradio/gnuradio/blob/4722ef5b2894be5bc69ddf669232be6eba1b0cca/gr-utils/blocktool/core/parseheader_generic.py#L322

removing `define_symbols=self.define_symbols` to `define_symbols='',`

I think that perhaps some better handling of defines if they are verily zero should be enabled here. I think what my defines is at the moment is a tuple `('',)`

### System Information

OS: 24.04
GR Installation Method: APT

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.9.2

### Steps to Reproduce the Problem

Try and build an OOT module with no defines. 

### Relevant log output

```shell
See above
```

PeterQFR, 2025-03-19:
Just some more steps to reproduce the problem. On Line322, replace `self.define_symbols` with `("",)` and it will error. if you use `("")` (ie without trailing comma) it will work. `("", "")` will error with
```
<command line>:1:9: error: macro name must be an identifier
    1 | #define -D 1

```

marcusmueller, 2025-03-22:
> I think that perhaps some better handling of defines if they are verily zero should be enabled here. I think what my defines is at the moment is a tuple ('',)

I'm not 100 % certain I follow you here, but I think you are proposing:

```python
    define_symbols=self.symbol_defines(),
…

  def symbol_defines():
    # check whether define_symbols is empty, or
    # none of its elements are anything but empty
    if (not self.define_symbols) or not any(self.define_symbols):
      return ""
    return " ".join(self.define_symbols)
```

or something similar?

PeterQFR, 2025-03-22:
Yes, something like that may work and is probably the best. Alternatively is to change the call to the parser to omit setting defines if they are not available, but that doubles up code (with only a change to the call), this is probably better.

marcusmueller, 2025-03-23:
@PeterQFR  thank you!!