gnuradio/gnuradio #7783: Debug tools for Audio (Sink|Source) device selection

csylvain, 2025-03-23:
### Feature Description

when unexpected or confusing results occur in GRC regarding selecting audio devices in gr-audio (particularly when using the <ins>windows</ins> or <ins>portaudio</ins> backends), the radioconda environment does not provide much in the way of debugging tools to ascertain what is going wrong other than debug log entries at runtime.

### Feature Urgency

high (impacts broad portion of ecosystem)

### More Information

[in reference to `gr-audio` in **Gnuradio Companion 3.10.10.0 (Python 3.11.9) installed via radioconda**]

when using the <ins>portaudio</ins> backend, including the sounddevice python package would be extremely helpful. troubleshooting when this package is installed by simply doing `python -m sounddevice` in the conda environment.

presently, one must `pip install --user sounddevice` to add the package to the conda environment to enable being able to produce the enumeration and names of audio devices by the <ins>portaudio</ins> backend.

when using the <ins>windows</ins> backend, troubleshooting documentation is necessary, as tools external to the conda environment are required.

in an elevated Powershell, one must `Install-Module -Name AudioDeviceCmdlets` to add debug commands to the Powershell environment. subsequently, using a regular unelevated Powershell, the command `get-audiodevice -list` produces an enumeration and names of audio devices available to the <ins>windows</ins> backend.

[this Feature Request is related to a forthcoming Issue reporting a bug in `gr-audio` when <ins>auto</ins> backend selection is enabled]

ryanvolz, 2025-03-25:
> presently, one must `pip install --user sounddevice` to add the package to the conda environment to enable being able to produce the enumeration and names of audio devices by the portaudio backend.

The conda-forge package is `python-sounddevice`, which would be preferable to pip install for the sake of maintaining a consistent environment. It would be only a small effort to include that in radioconda, and with some documentation additions about how to query for audio devices that would be enough to improve things in the short term. I like the discussions in the related issues about longer-term solutions.

csylvain, 2025-03-25:
tried that before using `pip` in the conda environment.
```
conda install sounddevice underway ..

PackagesNotFoundError: The following packages are not available from current channels:

    sounddevice
```

or was the package not found because `sounddevice` != `python-sounddevice` ?

ryanvolz, 2025-03-25:
> or was the package not found because `sounddevice` != `python-sounddevice` ?

Correct, it needs to be `python-sounddevice`. Package naming between PyPI and conda-forge can be tricky, because conda packages don't have to be related to Python and where there could be ambiguity `python-` often gets prefixed.