gnuradio/gnuradio #7784: inconsistent backend selection in `gr-audio` when <ins>auto</ins> selection mode is enabled

csylvain, 2025-03-23:
### What happened?

[in reference to `gr-audio` in **Gnuradio Companion 3.10.10.0 (Python 3.11.9) installed via radioconda**]

in `gr-audio.conf` the backend module selection is presently set to <ins>auto</ins> in the Windows conda release
```
[audio]

audio_module = auto
```

what happens next is inconsistent selection of audio backends for Sinks versus Sources when using `gnuradio-companion`
```
audio_source :info: Audio source arch: portaudio
audio_portaudio_source (debug) :info: Test Devices
audio_portaudio_source (debug) :info: Testing device name: Microsoft Sound Mapper - Input...
audio_portaudio_source (debug) :info: Testing device name: Analogue 1 + 2 (Focusrite USB A...
audio_portaudio_source (debug) :info:   Chosen!
audio_portaudio_source (debug) :info: Analogue 1 + 2 using MME as the host
```
versus
```
audio source :info: Audio sink arch: windows
audio_windows_sink :warning: waveOut device 'CABLE-B Input' was not found, defaulting to WAVE_MAPPER
audio_windows_sink (debug) :info: Opened windows waveout device
audio_windows_sink (debug) :info: Initialized 4 100 ms audio buffers, total memory used: 75.00 kB
```

this is extremely confusing and unexpected behavior. expected behavior is consistent backend selection.

# Workaround

manually select a backend for `gr-audio`, which forces consistent behavior by the conda release in Windows.
```
[audio]

#audio_module = auto
audio_module = portaudio
```

### System Information

OS: Windows 11 24H2
GR Installation Method: radioconda

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.10.0

### Steps to Reproduce the Problem

see **What happened?** narrative

### Relevant log output

```shell

```

marcusmueller, 2025-03-23:
Thanks for the excellent report here and in chat!

ryanvolz, 2025-03-25:
I did some searching since I recalled a time when the Windows backend was lowered in priority to below portaudio, figuring it would be a place to start: https://github.com/gnuradio/gnuradio/pull/4107

But there I saw a comment that makes me look like Nostradamus, and points to what's happening: 
> Does the Windows sink also need to be changed to low priority?

So the fix should be straightforward. 😁

marcusmueller, 2025-03-25:
@ryanvolz you're now the official Oracle of GNU Radio. It's less of a "Oracle of Delphi" kind of a deal (this is a C++/Python project, after all), and probably much more Cassandra-ian, i.e., you make such comments, and have me come back half a decade later and acknowledge you've been right ;-)

PR #7790 (2025-06-10): audio: unify priorities between sink and source architectures
## Description
I made a mistake 4 years ago that gave the "windows" sink a different
priority than the source. That led to great user confusion.

Make sure that doesn't happen again. Instead, keep priorities in a
central map.

Remove the internal and unused registry functions that just create a struct manually instead of using the aggregate initialization method / default constructor.

## Related Issue

Fixes #7784

## Which blocks/areas does this affect?

inner workings of gr-audio. Should un-shamble windows.

## Testing Done

work [... cut]