gnuradio/gnuradio #7785: add "list sinks / sources" functionality to audio backends, make available

marcusmueller, 2025-03-23:
### Feature Description

Users should be able to get a list of valid audio devices to put into their audio sink source field.

### Feature Urgency

medium (would be nice to have in the near future)

marcusmueller, 2025-03-23:
Slightly complicated by the architecture of gr-audio's backends. 

I think we need to first implement a `jack://sinkname` /  `alsa://sinkname` /  `windows://sinkname` / … mechanism to allow users explicitly select the audio system via the device name field. A nice idea would be a QT GUI widget for runtime selection of sound system and device.

csylvain, 2025-03-24:
i would expect then the default would be `auto://` in the Device Name field, explicitly or implicitly.

i propose it would be sufficient for GRC to present the ability to provide information about the environment in which it is running by entries in the main menu Tools entry.  and while considering that, another Tools entry to provide the `gnuradio-config-info` informaiton would be very helpful, indeed.

a GUI selector activated by poking around in the Device Name field would be nice, but maybe not necessary.

marcusmueller, 2025-03-25:
Thing is that GRC is pure-python, no-interaction-with-the-audio-system, so pulling in a dependency on any sound systems there feels much less desirable (and much more maintenance work) than letting the actual places where we already talk to the sound systems expose an API. 

Hence, if GRC gets any of that capability, it would be through an abstraction layer exposed by gr-audio (and that's what's proposed by me).

csylvain, 2025-03-25:
even pure-python GRC can invoke the new *spit out a list of every device you find* part of `gr-audio`, can't it?
same with invoking the *tell me all about the runtime environment* part(s) of `gnuradio-config-info`, yes?

marcusmueller, 2025-03-28:
@csylvain yes. The point of this issue that someone needs to write that new gr-audio functionality.