gnuradio/gnuradio #7791: Failed to build tecla on  MAC M1

cn0xroot, 2025-03-28:
### What happened?

A bug happened!

```
sudo port install gr-osmosdr
--->  Computing dependencies for gr-osmosdr
The following dependencies will be installed:
 bladeRF
 groff
 help2man
 netpbm
 p5.34-locale-gettext
 pandoc
 psutils
 tecla
 uchardet
Continue? [Y/n]: y
--->  Fetching archive for tecla
--->  Attempting to fetch tecla-1.6.3_0.darwin_24.arm64.tbz2 from https://mirror.sjtu.edu.cn/macports/packages/tecla
--->  Attempting to fetch tecla-1.6.3_0.darwin_24.arm64.tbz2 from https://pek.cn.packages.macports.org/macports/packages/tecla
--->  Attempting to fetch tecla-1.6.3_0.darwin_24.arm64.tbz2 from https://kmq.jp.packages.macports.org/tecla
--->  Building tecla
Error: Failed to build tecla: command execution failed
Error: See /opt/local/var/macports/logs/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_devel_tecla/tecla/main.log for details.
Error: Follow https://guide.macports.org/#project.tickets if you believe there is a bug.
```

```
port info gr-osmosdr
gr-osmosdr @20210117-cffef690_3 (science, comms)
Sub-ports:            gr37-osmosdr
Variants:             [+]airspy, [+]airspyhf, [+]bladeRF, debug, [+]docs, [+]fcdproplus, [+]hackrf, iio, python310, python38, [+]python39, [+]redpitaya, [+]rfspace,
                      [+]rtlsdr, sdrplay, [+]soapysdr, [+]spyserver, [+]uhd, universal

Description:          Includes OSMO SDR support GNU Radio source and sink blocks in C++, Python, and GNU Radio Companion (grc). This port also offers a wrapper
                      functionality for the FunCube Dongle, FunCube Dongle Pro+, HackRF, RFSpace, Ettus UHD, and rtl-sdr radios and software. By using gr-osmosdr source
                      you can take advantage of a common software API in your applications independent of the underlying radio hardware.
Homepage:             http://sdr.osmocom.org/trac/wiki/GrOsmoSDR

Build Dependencies:   cmake, pkgconfig, swig-python, doxygen, graphviz, py39-sphinx, texlive-latex
Library Dependencies: fftw-3-single, mpir, volk, gmp, python39, py39-numpy, py39-cheetah, gnuradio, uhd, hackrf, gr-fcdproplus, rtl-sdr, bladeRF, airspy, SoapySDR,
                      airspyhf, boost171
Runtime Dependencies: py39-matplotlib, py39-opengl, py39-scipy
Conflicts with:       gr37-osmosdr
Platforms:            darwin
License:              GPL-3
Maintainers:          Email: michaelld@macports.org, GitHub: michaelld
                      Policy: openmaintainer
```

 cc  @michaelld  

```
port installed  |grep gr
  gnuradio @3.8.5.0_3+docs+grc+python39+qtgui+sdl+uhd+wavelet+zeromq (active)
  gr-fcdproplus @20200811-575fcafb_0+docs+python39 (active)
  graphite2 @1.3.14_0 (active)
  graphviz @12.2.1_0+pangocairo+x11 (active)
  py39-pyqtgraph @0.13.7_0+opengl+pyqt5 (active)
  qt5-qtgraphicaleffects @5.15.15_0+examples (active)
```

### System Information

OS: (Mac M1)
GR Installation Method: ( mac port

### GNU Radio Version

3.8  (maint-3.8)

### Specific Version

 gnuradio-config-info -v 3.8.5.0

### Steps to Reproduce the Problem

```
sudo port install tecla
--->  Computing dependencies for tecla
--->  Fetching archive for tecla
--->  Attempting to fetch tecla-1.6.3_0.darwin_24.arm64.tbz2 from https://mirror.sjtu.edu.cn/macports/packages/tecla
--->  Attempting to fetch tecla-1.6.3_0.darwin_24.arm64.tbz2 from https://pek.cn.packages.macports.org/macports/packages/tecla
--->  Attempting to fetch tecla-1.6.3_0.darwin_24.arm64.tbz2 from https://kmq.jp.packages.macports.org/tecla
--->  Building tecla
Error: Failed to build tecla: command execution failed
Error: See /opt/local/var/macports/logs/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_devel_tecla/tecla/main.log for details.
Error: Follow https://guide.macports.org/#project.tickets if you believe there is a bug.
Error: Processing of port tecla failed
```

### Relevant log output

```shell

```

cn0xroot, 2025-03-28:
More info :

System: 15.3.2
Xcode: 16.2
Chip:Apple M1 Ultra

marcusmueller, 2025-03-28:
Hi! We're not the maintainers of gr-osmosdr, nor of the MacPort build; I'd recommend talking to the MacPorts folks.

Also, we don't really maintain GNU Radio 3.8.x.y anymore; the hardware you're running it on didn't even exist when we released that :)

closed 2025-03-28 by marcusmueller