gnuradio/gnuradio #7795: GRC Qt: It's possible to delete the Options block

haakov, 2025-03-30:
### What happened?

It's possible to delete the Options block in GRC Qt. It should be impossible to delete, like in GRC Gtk.

### System Information

OS: Linux
GR Installation Method: Source

### GNU Radio Version

3.11-git (main)

### Steps to Reproduce the Problem

Open GRC Qt, select the Options block, hit Delete

### Relevant log output

```shell

```

stephenm7777, 2025-03-31:
i was not able to reproduce this

marcusmueller, 2025-03-31:
@stephenm7777 I definitely can, and considering  @haakov wrote that piece of software: are you sure you're running the Qt version of GRC?

stephenm7777, 2025-04-28:
Apologies @marcusmueller as I have been busy but I ran it exactly as it says on the documentation 
`How to Run the QT Version of GNU Radio Companion
    # Ensure PyQt5 and other dependencies are installed
    pip install QDarkStyle qtpy
    
    # If you want to run tests
    pip install pytest-qt pyautogui

    # Build and install as usual (described below) and then run
    gnuradio-companion --qt`

has the documentation not been updated or is there another way of running to QT version?

marcusmueller, 2025-04-29:
No, that's the right way to do it :) OK, interesting that it can't be deleted on your machine! That is a surprising inconsistency, rendering this bug even more interesting.