gnuradio/gnuradio #7818: blocks: peak_detector gets wrong peak or fails to find peak

ffu, 2025-04-26:
### What happened?

When testing block peak_detector_fb with a raised cosine wave y[n] = 1 + cos(n*2*$\\pi$/10), I expect peaks to occur at indices 10*k, where k is an integer. Occasionally, either the detected peak is off by 1, or no peak is found.

Attached are screenshots of both cases.

[image]

[image]

Blue line is the peak detector input. Orange line is the peak detector output.

Analysis shows problem occurs in peak_detector_impl.cc:
```
    if (state == 0) {
        // printf("Leave in State 0, produced %d avg %f\n",noutput_items, d_avg);
        return noutput_items;
    } else { // only return up to passing the threshold
        // printf("Leave in State 1, only produced %d of %d %f\n",peak_ind,noutput_items, d_avg);
        return peak_ind + 1;
    }
}
```
During the work function, if the state == 1, then returning peak_ind + 1 will set the output corresponding to the peak index to 0 (no peak) and this local peak in the input buffer is consumed. Therefore, an attempt will be made to find the local peak in the next call to work. In the next call to work, if there are enough samples, another peak could be found at a different location. Sometimes, due to the throttle block, the next noutput_items is 1. In that case, the input is consumed again, and the next sample may be beneath the fall threshold, resulting in no peak output.

This block is short but complicated due to it's multi-state implementation while attempting to keep track of the average. I attempted to fix by changing to "return peak_ind". This will solve the peak calculation, but this could also create repeated unconsumed loops, especially when noutput_items is 1 and the work function returns 0. In a similar fashion, if the input is a plateau or step function, that change may cause the block to become stuck. Therefore, a suitable fix may require some significant changes. As it stands, this investigation shows that peak_detector2 may be a better alternative since it truly implements the look_ahead parameter.

### System Information

OS: Linux and Mac
GR Installation Method: Source (Linux) and Homebrew (Mac)

### GNU Radio Version

3.11-git (main)

### Specific Version

g884e8f6dd

### Steps to Reproduce the Problem

```
from gnuradio import blocks, gr, analog
import matplotlib.pyplot as plt
import numpy as np

THROTTLE_RATE = 1e6 # Increase throttle rate to see the off-by-one error
NUMSAMPS = 9000
SAMP_FREQ = 10
tb = gr.top_block()
pkdet = blocks.peak_detector_fb(1.5, 1.5)
cos = analog.sig_source_f(SAMP_FREQ,
                          analog.GR_COS_WAVE,
                          wave_freq=1,
                          ampl=1,
                          offset=1)

snk_peak = blocks.vector_sink_b()
snk_cos = blocks.vector_sink_f()
head = blocks.head(4, NUMSAMPS)
throttle = blocks.throttle(4, THROTTLE_RATE)
tb.connect(cos, throttle, head, pkdet, snk_peak)
tb.connect(head, snk_cos)
tb.run()

out = np.array(snk_peak.data())
inpt = np.array(snk_cos.data())

expected_peak_indices = range(0, len(out), SAMP_FREQ)
errors = [n for n in expected_peak_indices if out[n] != 1]
if errors:
    for e in errors:
        print("No peak found at index", e)
    first_error = errors[0]
    x = np.arange(first_error-20, first_error + 21)
    plt.plot(x, inpt[x])
    plt.plot(x, out[x])
    plt.show()
```

### Relevant log output

```shell

```

marcusmueller, 2025-04-26:
Thank you for this excellent bug report!

marcusmueller, 2025-04-26:
Discussion from chat with bug reporter: Yes. We should **deprecate this block**.