gnuradio/gnuradio #7820: Modtool: shell_complete is deprecated

martin-feaux, 2025-04-30:
### What happened?

Modtool throw an exception: see `TypeError: __init__() got an unexpected keyword argument 'shell_complete'`.
This happen while trying to use modtool (see relevant log output).

This is due to a change in Click, which deprecated `shell_complete`. See https://github.com/fastapi/typer/pull/974.

The easy fix is to replace `shell_completion` by `autocompletion`.

```
diff --git a/gr-utils/modtool/cli/base.py b/gr-utils/modtool/cli/base.py
--- a/gr-utils/modtool/cli/base.py
+++ b/gr-utils/modtool/cli/base.py
@@ -160,7 +160,7 @@ def block_name(**kwargs):
         )
     return click.argument(
         'blockname', nargs=1, required=False, metavar="BLOCK_NAME",
-        shell_complete=block_name_complete)
+        autocompletion=block_name_complete)
 
 
 @click.command(cls=CommandCLI,
```

There are others place where `shell_complete` is used:

 ```
$ grep -rn "shell_complete" .
./gr-utils/modtool/cli/update.py:31:@click.argument("blockname", nargs=1, required=False, metavar="BLOCK_NAME", shell_complete=xml_blockname_complete)
./gr-utils/modtool/cli/add.py:32:    def shell_complete(self, ctx, param, incomplete: str):
```

This issue probably depends on Click version (see system information for mine).

### System Information

OS: Debian 11
GR Installation Method: Source (22bc9e07fbccff63fd1d7b061587858bd188b772)

```
$ python3
Python 3.9.2 (default, Dec  1 2024, 12:12:57) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import click
>>> print(click.__version__)
7.1.2
```

### GNU Radio Version

3.11-git (main)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

In a OOT, use `gr_modtool bind custom_block`

### Relevant log output

```shell
$ gr_modtool bind custom_block
Traceback (most recent call last):
  File "/usr/local/bin/gr_modtool", line 18, in <module>
    cli()
  File "/usr/lib/python3/dist-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1254, in invoke
    cmd_name, cmd, args = self.resolve_command(ctx, args)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1297, in resolve_command
    cmd = self.get_command(ctx, cmd_name)
  File "/usr/local/lib/python3.9/dist-packages/gnuradio/modtool/cli/base.py", line 76, in get_command
    mod = import_module('gnuradio.modtool.cli.' + cmd_name)
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/usr/local/lib/python3.9/dist-packages/gnuradio/modtool/cli/bind.py", line 39, in <module>
    def cli(**kwargs):
  File "/usr/lib/python3/dist-packages/click/decorators.py", line 168, in decorator
    _param_memo(f, ArgumentClass(param_decls, **attrs))
  File "/usr/lib/python3/dist-packages/click/core.py", line 1984, in __init__
    Parameter.__init__(self, param_decls, required=required, **attrs)
TypeError: __init__() got an unexpected keyword argument 'shell_complete'
```

marcusmueller, 2025-04-30:
This is a bit confusing. In [Click 8.1.0](https://click.palletsprojects.com/en/stable/changes/#version-8-1-0) they renamed "autocompletion" to "shell_complete", and now you propose going the opposite direction, because a different library deprecates "shell_complete". 

The  stable API doesn't list `autocompletion` at all, so your fix couldn't work with a stable Click release, or am I mistaken here?

martin-feaux, 2025-04-30:
Indeed, seems like I connected unexisting dot.
Real solution would be to upgrade click on my side ?
I thought I add the last version since I downloaded yesterday all dependencies following this link: https://wiki.gnuradio.org/index.php?title=UbuntuInstall

marcusmueller, 2025-04-30:
@martin-feaux yes, that's possible. It would be as easy, though, to just add a compat layer. 

Generally, we do not support Debian 11 anymore for new software – and any fix you'd get would be due to new software – so, I'm not sure whom that addition would help! Only people manually building a modern "alive" GNU Radio on a "dead" platform would benefit from that.

I'll discuss internally and let you know.

marcusmueller, 2025-04-30:
(on a serious remark: you're really using an end-of-life version of debian. The right thing to do is to update your debian, everything else nonwithstanding. You're a security risk to yourself, and upgrading debian is not that hard, actually.)

closed 2025-04-30 by marcusmueller