gnuradio/gnuradio #7826: GRC Qt: Properties dialog crashes for UHD USRP Sink|Source  blocks

sl077, 2025-05-14:
### What happened?

I installed a current gnuradio from conda-forge in a conda environment to work with URSPs. To roughly test the setup, I opened GRC with the Qt GUI, added an "UHD: USRP Sink" block and did a doubleclick on it. This crashed the GRC.

Console output shows "ValueError: '-1.0' is not in list" in line 153 of grc/gui_qt/components/dialogs.py:
https://github.com/gnuradio/gnuradio/blob/22bc9e07fbccff63fd1d7b061587858bd188b772/grc/gui_qt/components/dialogs.py#L153
Further debugging shows, that the parameter "start_time" in uhd_usrp_sink.block.yml (+ uhd_usrp_source.block.yml) is causing this, where above code seems to search for a string in a list of floats (actually a single one: -1.0).

### System Information

OS: Linux Debian Trixie
GR Installation Method: Conda

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.12.0

### Steps to Reproduce the Problem

Open the Properties dialog of the "UHD: USRP Sink" or "UHD: USRP Source" block. GRC should crash with a ValueError.

### Relevant log output

```shell
Traceback (most recent call last):
  File "/opt/modules/envs/conda/20250509_gnuradio/lib/python3.12/site-packages/gnuradio/grc/gui_qt/components/canvas/block.py", line 398, in mouseDoubleClickEvent
    self.open_properties()
  File "/opt/modules/envs/conda/20250509_gnuradio/lib/python3.12/site-packages/gnuradio/grc/gui_qt/components/canvas/block.py", line 428, in open_properties
    self.props_dialog = PropsDialog(self.core, self.force_show_id)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/modules/envs/conda/20250509_gnuradio/lib/python3.12/site-packages/gnuradio/grc/gui_qt/components/dialogs.py", line 85, in __init__
    qvb = self.build_param_entrys(cat)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/modules/envs/conda/20250509_gnuradio/lib/python3.12/site-packages/gnuradio/grc/gui_qt/components/dialogs.py", line 153, in build_param_entrys
    dropdown.param_values.index(param.get_value())
ValueError: '-1.0' is not in list
Aborted
```

ergobound, 2025-09-10:
Same problem. Any solutions, dear friends?

```
Traceback (most recent call last):
  File "C:\ProgramData\radioconda\Lib\site-packages\gnuradio\grc\gui_qt\components\canvas\block.py", line 398, in mouseDoubleClickEvent
    self.open_properties()
  File "C:\ProgramData\radioconda\Lib\site-packages\gnuradio\grc\gui_qt\components\canvas\block.py", line 428, in open_properties      
    self.props_dialog = PropsDialog(self.core, self.force_show_id)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\ProgramData\radioconda\Lib\site-packages\gnuradio\grc\gui_qt\components\dialogs.py", line 85, in __init__
    qvb = self.build_param_entrys(cat)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\ProgramData\radioconda\Lib\site-packages\gnuradio\grc\gui_qt\components\dialogs.py", line 153, in build_param_entrys
    dropdown.param_values.index(param.get_value())
ValueError: '1' is not in list
```

Windows; Gnu Radio 3.10.12.0; Python 3.12.9

Anjali503, 2026-01-14:
I’ve opened PR #8008 to fix this by normalizing dropdown values and parameter
values before indexing, which avoids the type-mismatch crash in the GRC Qt
Properties dialog.

lychaxo, 2026-09-12:
I encountered the same issue on GNU Radio 3.10.12.0 (Python 3.9.10, Qt 5.15.3), by opening examples/soapy/fm_radio_receiver_soapy.grc and then double-clicking Audio Sink (which has Variable audio_rate entered for Sample Rate) to edit the properties.  I tried the above workaround https://github.com/gnuradio/gnuradio/issues/7826#issuecomment-3272824745 but the crash still happened.
```
Traceback (most recent call last):
  File "/usr/lib64/python3.9/site-packages/gnuradio/grc/gui_qt/components/canvas/block.py", line 398, in mouseDoubleClickEvent
    self.open_properties()
  File "/usr/lib64/python3.9/site-packages/gnuradio/grc/gui_qt/components/canvas/block.py", line 428, in open_properties
    self.props_dialog = PropsDialog(self.core, self.force_show_id)
  File "/usr/lib64/python3.9/site-packages/gnuradio/grc/gui_qt/components/dialogs.py", line 85, in __init__
    qvb = self.build_param_entrys(cat)
  File "/usr/lib64/python3.9/site-packages/gnuradio/grc/gui_qt/components/dialogs.py", line 160, in build_param_entrys
    dropdown.param_values.index(value)
ValueError: 'audio_rate' is not in list
Aborted
```

The updated PR #8008 from @Anjali503 doesn't seem to work as I would expect either; the variable audio_rate is replaced with the first enum value.  I got this more or less working, to auto-populate new options (e.g. if you enter -0.5 or my_start_time for the USRP Sink start_time, or my_new_audio_rate for the soapy example):
```
--- grc/gui_qt/components/dialogs.py	2026-09-12 10:44:28.788137330 -0400
+++ /usr/lib64/python3.9/site-packages/gnuradio/grc/gui_qt/components/dialogs.py2026-09-12 12:45:59.252706317 -0400
@@ -1,5 +1,5 @@
 from __future__ import absolute_import, print_function
-
+import pprint # just for debugging the {new code}
 from copy import copy
 
 from ..Constants import MIN_DIALOG_HEIGHT, DEFAULT_PARAM_TAB
@@ -135,23 +135,59 @@
                     dtype_label = QLabel(f"[{param.dtype}]")
                 qvb.addWidget(QLabel(param.name), i, 0)
                 if param.dtype == "enum" or param.options:
-                    dropdown = QComboBox()
-                    for opt in param.options.values():
-                        dropdown.addItem(opt)
-                    dropdown.param_values = list(param.options)
-                    dropdown.param = param
-                    qvb.addWidget(dropdown, i, 1)
-                    self.edit_params.append(dropdown)
-                    if param.dtype == "enum":
-                        dropdown.setCurrentIndex(
+                    dropdown = QComboBox() # new dropdown
+                    for opt in param.options.values(): # dict options ->
+                        dropdown.addItem(opt) # populate widget
+                    dropdown.param_values = list(param.options) # widget data
+                    dropdown.param = param # widget map to param
+                    qvb.addWidget(dropdown, i, 1) # add to panel
+                    self.edit_params.append(dropdown) # trigger connection
+                    if param.dtype == "enum": # for enum
+                        dropdown.setCurrentIndex( # widget select current
                             dropdown.param_values.index(param.get_value())
                         )
                         dropdown.currentIndexChanged.connect(self.gui_update)
-                    else:
-                        dropdown.setEditable(True)
-                        dropdown.setCurrentIndex(
-                            dropdown.param_values.index(param.get_value())
-                        )
+                    else: # for option
+                        dropdown.setEditable(True) # editable, and appendable:
+                        dropdown.setInsertPolicy(QComboBox.InsertPolicy.InsertAtBottom)
+
+                        if(param.key == "samp_rate" or param.key == "start_time"):
+                            print("key=%r name='%r' get_value()=%r options:" % (
+                                param.key, param.name, param.get_value()
+                            ))
+                            print(vars(param.options))
+                            print("> options.items():")
+                            pprint.pp(param.options.items())
+
+                        # https://github.com/gnuradio/gnuradio/issues/7826#issuecomment-3272824745
+                        # avoids crash for type mapping problem but doesn't handle variable mapping
[... 47 lines cut ...]
+                    if str(current_val) != user_text: # see if new entry
+                        print("* new '%r' (not '%r')" % (user_text, str(current_val)))
+                        try:
+                            target_type = type(current_val) # get type
+                            print("* * casting to %r'" % target_type)
+                            user_val = target_type(user_text) # try to cast it
+                            par.param.options[user_text] = user_val # new opt
+                            par.param.set_value(user_text) # choose new opt
+                            if hasattr(par, 'param_values') and user_val not in par.param_values:
+                                print("* adding %r to:" % user_val)
+                                pprint.pp(par.param_values)
+                                par.param_values.append(user_val) # just in case
+                        except (ValueError, TypeError):
+                            print("*** could not cast '%r'" % user_text)
+                            pass # bad entry from user
[... cut ...] x.

Excuse all my debug prints -- I left them in as it may be helpful to understand what I'm attempting here.   I don't know PyQt at all nor have a very good understanding of grc's implementation (maybe someday, just don't have the time right now!).
If a nonexistant variable is entered, it is handled more or less how you would expect (dialog error pane shows up telling you about invalid value), but I do wonder if the except/pass earlier needs more than just giving up silently (though maybe not).