gnuradio/gnuradio #7829: Plots (QT GUI Time Sink) are very slow and laggy on GNU Radio 3.10.12 via Radioconda on macOS

MatheusAlves65, 2025-05-19:
### What happened?

When running a simple flowgraph in GNU Radio 3.10.12 installed via Radioconda on macOS, the GUI plots (such as QT GUI Time Sink) update very slowly, causing lag and high CPU usage.

I expected the plots to update smoothly and responsively, without excessive CPU consumption.

### System Information

OS: macOS (MacOS Sequoia 15.5)
GNU Radio Installation Method: Conda (Radioconda environment)
GNU Radio Version: 3.10.12.0

### GNU Radio Version

4.0-git (dev-4.0)

### Specific Version

3.10.12.0

### Steps to Reproduce the Problem

Create a simple flowgraph with a Signal Source connected to a QT GUI Time Sink block in GNU Radio Companion.

Run the flowgraph without adding a Throttle block.

Observe that the plot updates very slowly and the interface becomes laggy.

### Relevant log output

```shell
Warning: This flow graph may not have flow control: no audio or RF hardware blocks found. Add a Misc->Throttle block to your flow graph to avoid CPU congestion.

log :warning: Matplotlib is a required dependency to use DistanceRadar and AzElPlot. Please install matplotlib to use these blocks.
```

dkozel, 2025-05-28:
High CPU usage is expected in a flowgraph without the Throttle block or a hardware Source/Sink such as an audio soundcard or radio. GNU Radio's scheduler will run the data flow at maximum possible speed, limited only by the CPU's resources as noted in the warning in the log.

The QT GUI displays I believe will lag if handling too much data. It is known that the data handling within them could be improved significantly, but they do handle up to several hundred MS/s reasonably well. If that is not the case on MacOS with Conda it would be a good occasion to debug the issue and make at least an incremental improvement.

khimaros, 2025-06-04:
it does appear that the performance of QT GUI Time Sink, in particular, has fallen since GNURadio 3.10.4.0. i'm using a hackrf source.

i have a number of flow graphs from that version which were very performant on the same hardware and now completely freeze the UI.

disabling the plots or setting `Number of Points` to something small (less than 5k) seems to be the only way to make the UI responsive.

previously, i was able to get away with closer to 130k points.

AndersBNielsen, 2025-08-21:
Just installed 3.10.12.0_2 via brew. Even running the Sample Rate Tutorial is slow, and eventually completely unresponsive. (Including throttle of course)
Something is completely off at the moment. 

https://wiki.gnuradio.org/index.php/Sample_Rate_Tutorial

therokh, 2026-09-30:
Installed 3.10.12.0 on Fedora 44 and seeing the same issue. The plots are very slow to update and the time series plot can sometimes freeze for a few seconds at a time. This was from visualizing a 1k waveform at 32k samples.

marcusmueller, 2026-10-01:
@therokh this sounds like a different issue. Wild guess: there's a Throttle in your flowgraph, and you're *not* using the "limiting" parameters of said throttle. If that's not the case, please open a new bug report, because this one is only about mac os.