gnuradio/gnuradio #7830: Add a Vector Slice block

dkozel, 2025-05-21:
### Feature Description

The Vector Slice block has three parameters:
* Input Vector Length
* Output Vector Length
* Start Index

Asserts that Output Length <= Input Length and Start Index + Output Length <= Input Length

### Feature Urgency

low (just an idea)

marcusmueller, 2025-05-21:
so, a less useless-copies-version of vector to stream -> keep m in n -> stream to vector. I like!

TheWylieStCoyote, 2025-07-01:
How about Numpy slice notation?

jsrv07, 2025-12-05:
Hi @dkozel and @marcusmueller,
I'd like to take this issue if it's still available.My plan is to implement it as a gr::sync_block.

Approach
In the constructor, I’ll check that output_len <= input_len and that start_idx + output_len does not exceed input_len.
The I/O signatures would be input_len * sizeof(type) for the input and output_len * sizeof(type) for the output.
In work(), I'll simply slice/copy the required portion of the input buffer into the output buffer.

Let me know if this direction makes sense or if you have any suggestions before I start coding.